gopcua/opcua #318: Any example for monitoring with param "Filter"

nbfhscl, 2020-02-14:
I've tried the following codes but ProsysOpcuaServer(4.0.2-108) returns "The monitored item filter parameter is not valid. StatusBadMonitoredItemFilterInvalid (0x80430000)". 
```
			request.RequestedParameters.Filter = ua.NewExtensionObject(
				ua.DataChangeFilter{
					Trigger: ua.DataChangeTriggerStatus,
				},
			)
```

magiconair, 2020-02-18:
No, we don't have an example for filters. Try registering the `DataChangeFilter` in `ua/extension_object.go`. Right now we're not registering all default ones. 

@kung-foo #277

nbfhscl, 2020-02-19:
@magiconair Thanks. The following code seems working.
`case "Status":
			request.RequestedParameters.Filter = &ua.ExtensionObject{
				EncodingMask: ua.ExtensionObjectBinary,
				TypeID:       ua.NewFourByteExpandedNodeID(0, id.DataChangeFilter_Encoding_DefaultBinary),
				Value: ua.DataChangeFilter{
					Trigger: ua.DataChangeTriggerStatus,
				},
			}
		case "Value":
			request.RequestedParameters.Filter = &ua.ExtensionObject{
				EncodingMask: ua.ExtensionObjectBinary,
				TypeID:       ua.NewFourByteExpandedNodeID(0, id.DataChangeFilter_Encoding_DefaultBinary),
				Value: ua.DataChangeFilter{
					Trigger: ua.DataChangeTriggerStatusValue,
				},
			}
		case "Timestamp":
			request.RequestedParameters.Filter = &ua.ExtensionObject{
				EncodingMask: ua.ExtensionObjectBinary,
				TypeID:       ua.NewFourByteExpandedNodeID(0, id.DataChangeFilter_Encoding_DefaultBinary),
				Value: ua.DataChangeFilter{
					Trigger: ua.DataChangeTriggerStatusValueTimestamp,
				},
			}`

magiconair, 2020-02-19:
Hmm, After registering the extension object you should not need to set these fields.

Can you try to add this to your code and then remove the `EncodingMask` and `TypeID` again?

```go
func init() {
    RegisterExtensionObject(NewNumericNodeID(0, id.DataChangeFilter_Encoding_DefaultBinary), new(DataChangeFilter))
}
```

nbfhscl, 2020-02-20:
I tried as you told, but it doesnt work for me, the server returns same error message. Is there any tips I missed or misunderstood.

```golang
func init() {
	ua.RegisterExtensionObject(ua.NewNumericNodeID(0, id.DataChangeFilter_Encoding_DefaultBinary), new(ua.DataChangeFilter))
}

request.RequestedParameters.Filter = ua.NewExtensionObject(
	&ua.DataChangeFilter{
		Trigger: ua.DataChangeTriggerStatus,
	},
)
```
By the way, manually adding EncodingMask and TypeID helps me succeed in monitoring, but the FilterResult filed returned in the MonitoredItemCreateResult seems always be nil, although the filter param did worked.

closed 2025-04-28 by github-actions[bot]