gopcua/opcua #330: OPC UA Events

flipback, 2020-03-14:
Hello everyone,

we are going to use your library in our projects, but one of our requirements is to OPCUA events. Does __opcua__ support this feature? If not (it seems it is not, because I can't find it in the sources), do you plan to integrate it?

Thank you.

magiconair, 2020-03-14:
Hi @flipback , I don't think so but if you can point me to the part of the spec then I can check how much work that would be. I had a quick look at Part 4 but couldn't find it there. 

PRs are of course welcome at any time.

flipback, 2020-03-14:
Hi @magiconair, thank you for the quick answer

Related specifications:

1. Spec 3: 4.6 Event Model
2. Spec 4: 7.17.3 EventFilter, 7.20.3 EventNotificationList parameter
3. Spec 5: 6.4.1 ObjectTypes used as EventTypes
4. Spec 9 - extends the event model (just for information)

Examples of servers:
 - https://open62541.org/doc/current/tutorial_server_events.html (C)
 - https://github.com/ASNeG/ASNeG-Demo/blob/master/src/ASNeG-Demo/Library/Event.cpp (C++)

Example of clients:
 - https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-events.py (Python)

I hope it can help to start.

Unfortunately, we can't provide a  PR with the feature, because we are short of time and resources. Sorry about it. But we can help to test it.

PR #349 (2020-06-18): Add example for event subscriptions and register 2 items for convenience
The code to allow node event subscriptions is already available in the existing codebase. This PR provides an example on how to subscribe to node events, and registers the event filter and event notification list nodes since they are returned when the event is triggered as a convenience so users do not need to register these themselves.

This PR should also resolve #330
closed 2020-06-18 by kung-foo