gopcua/opcua #331: Question on Monitoring Subscription

fdamador, 2020-03-16:
I have a question related to the monitoring PR  #252  Node Monitor (extended subscription).

I wanted to pass the msg.* values (id, value, server timestamp, client timestampt, quality) back to my main code from the MsgHandler, but I cannot interrogate the Subscription service for this data. 

Questions:
Is the nodeIds list and values stored int he subscrition service?  
Will this require a channel to pass this data back to the main subroutine?

https://github.com/gopcua/opcua/pull/252#issuecomment-599662073

kung-foo, 2020-03-16:
Did you follow the examples here: https://github.com/gopcua/opcua/blob/master/examples/monitor/monitor.go#L86-L90

Both callback and channel style return a `DataChangeMessage` (https://github.com/gopcua/opcua/blob/master/monitor/subscription.go#L29-L33) which have the node id and value(s).

fdamador, 2020-03-16:
thank you for the links. I will investigate them further.

All I want is to pass the values from the go-subroutines back to the main routines.

Perhaps, I'm not understanding the monitor->startCallbackSub() concurrent go routine.  I don't see how I can get to the msg data to the main routine without introducing a channel.

However, it looks like the startChanSub routine has the ChanSubscribe along with the sub.pump routines looks more of what I'm looking for.  It returns the *monitor.DataChangeMessage back from the routine it got called from.

closed 2020-03-22 by fdamador