gopcua/opcua #339: Cannot reconnect if connection is lost

jcrummy, 2020-04-08:
Part of my testing has been unplugging the connection between the client and the OPC server (in this case, they are one two different machines). Once a connection is lost, it can never connect again, even once the physical connection is restablished.

I've narrowed this down to line 143 in `client.go`:
[image]

This line checks that the secure channel is `nil`, otherwise it doesn't try to reconnect. Once a connection is established, the secure channel is never reset, even if you call `Close()` on the client.

I've worked around this for now my recreating my client any time the connection is lost (by calling `New (...)` and then calling `Connect()` again on this new client instance.

The secure channel should be able to be reconnected, however.

Jwangyi, 2020-05-29:
I have the same problem. Executive getValue()  per second,Once network disconnection, data will not lost . when the connection is recover. It  is still  wrong
```
func getValue(c *opcua.Client,req *ua.ReadRequest,ChannelSlice []ChannelInfo) (time.Time,time.Time,time.Time,time.Time,string){
	resp, err := c.Read(req)
	if err != nil {
		// log.Fatalf("Read failed: %s", err)
		fmt.Printf("%v",err)                       //**always print An existing connection was forcibly closed by the remote host**
		println(err)
		return f1,f2,f2,f2,"bad"
	}
            ...............
}

func doEvery(d time.Duration, c *opcua.Client,res *ua.ReadRequest,ChannelSlice []ChannelInfo,value1 string,db *sql.DB,f func(*opcua.Client, *ua.ReadRequest,[]ChannelInfo  ) (time.Time,time.Time,time.Time,time.Time,string)) {
	for x := range time.Tick(d) {
		name:="log.txt"          
		var content string
		var f1,f2,f3,endtime,sqls=f(c,res,ChannelSlice)
		if (sqls=="bad"){
			time1 := time.Now()
			content = time1.Format("2006-01-02 15:04:05")+"An existing connection was forcibly closed by the remote host\r\n"
			WriteWithFileWrite(name,content)
                }
                 ...........
        }
}
func main(){
	var (endpoint = flag.String("endpoint", "opc.tcp://10.242.132.34:49330", "OPC UA Endpoint URL"))
	ctx := context.Background()
	c := opcua.NewClient(*endpoint, opcua.SecurityMode(ua.MessageSecurityModeNone))
	if err := c.Connect(ctx); err != nil {
		fmt.Printf("ERR %v",err)
		// log.Fatal(err)
	}else{
		fmt.Printf("KEPW CONNECT Scuccess\n")
	}
       ...........
	go doEvery(1*time.Second,c,req,ChannelInfomaps[ChannelName],ChannelName,db,getValue);
}
```

fenogentov, 2021-11-16:
version 0.2.0
Problem still exists.
After client.Close ()
After disconnection

client.State() == Closed
debug: uasc Close()
opcua: already connected

henripqt, 2021-11-17:
Hi everybody,

I've also been having some trouble with this one. Here's a brief explanation of what happens on my side :

- I've got the prosys OPCUA Simulation server running
- I've subscribed to the default test objects available.
- I'm killing the simulation server in order to simulate a connection issue
- When reopening the simulation server I see my previous session is still there but my go application doesn't reconnect.

After investigating it turns out I receive the **`0x80790000 StatusBadNoSubscription` ** error (even though I still have subscriptions).

I've therefore added a case for handling the **`0x80790000 StatusBadNoSubscription` ** error in the client monitor error switch loop in order to set the action to `restoreSession` instead of defaulting to `createSecureChannel`. ATM this seems to work on my side.

```
[...]
default:
switch x := err.(type) {
case *uacp.Error:
	switch ua.StatusCode(x.ErrorCode) {
	case ua.StatusBadSecureChannelIDInvalid:
		// the secure channel has been rejected by the server
		action = createSecureChannel

	case ua.StatusBadSessionIDInvalid:
		// the session has been rejected by the server
		action = recreateSession

	case ua.StatusBadNoSubscription:
		action = restoreSession

	case ua.StatusBadSubscriptionIDInvalid:
	[...]
```

From my understanding this will restore the session along with the associated subscriptions if possible or recreate it however I don't understand yet why we're receiving the `StatusBadNoSubscription` error.

@magiconair Do you think this is the way to go  ? If so I'll continue my investigation and submit a PR

magiconair, 2021-11-17:
Don't know yet but this is good to know. Need to be able to reproduce this.

magiconair, 2021-11-19:
Hmm, I'm testing this with v0.2.2 and the `subscribe` example as follows and don't have an issue:

```sh
go run ./examples/subscribe/subscribe.go   -node 'i=2258' -mode None -interval 100ms -endpoint opc.tcp://better.local:53530/OPCUA/SimulationServer
```

magiconair, 2021-11-19:
It's kinda unhelpful that the only way to stop the simulation server is to terminate it.

magiconair, 2021-11-19:
I'm actually not seeing any session after I've started the Prosys Simulation server but the subscription keeps on working.

magiconair, 2021-11-19:
[image]

[image]

You can see it in the req/res log when you click "Filter by session". Yes, the session id is different. I think this is how the protocol is supposed to work since the simulation server doesn't remember the session anymore. Terminating the server is different from interrupting the connection.

magiconair, 2021-12-09:
I think these are actually three different problems. 

The original post was about the lack of auto-reconnect in v0.1.x. So recreating the client is the correct operation. 

The second issue from @fenogentov seems to be a different problem. Maybe you can explain this a bit more. 

The third issue from @henripqt is about not recreating subscriptions. 

I am going to close this issue and would ask the posters to create separate issues if the problem still persists. You can refer to this ticket.

@henripqt I would think that `restoreSubscription` should be the correct action during re-connect and it would be great if you can test that.

closed 2021-12-09 by magiconair