gopcua/opcua #340: Run on arm with Error

goodspeed1986, 2020-04-14:
Hello. I run my program on armv7 processor in linux and have this error. 
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x11da0]

goroutine 132 [running]:
runtime/internal/atomic.goXadd64(0x83cacc, 0x1, 0x0, 0x8922b8, 0x1)
        /home/user/go/src/runtime/internal/atomic/atomic_arm.go:103 +0x1c
github.com/gopcua/opcua/monitor.(*Subscription).pump(0x83cac0, 0x216eb0, 0x80ae40, 0x0, 0xb3c8d0)
        /root/go/src/github.com/gopcua/opcua/monitor/subscription.go:197 +0x138
created by github.com/gopcua/opcua/monitor.(*NodeMonitor).Subscribe
        /root/go/src/github.com/gopcua/opcua/monitor/subscription.go:111 +0xac
exit status 2

If i run it on my intel linux machine everything is fine. What am I doing wrong?

kung-foo, 2020-04-14:
Probably relevant: https://github.com/openvswitch/ovn-kubernetes/issues/454

goodspeed1986, 2020-04-14:
I seem to understand why it does not work, but I do not understand what should I do next. Maybe you have solutions to this problem?

goodspeed1986, 2020-04-14:
I changed in file /monitor/subscription.go 64 bit commands and variables to 32 bit and now it working in 32 bit arm processors.

kung-foo, 2020-04-14:
Yes, 32bit would work. But I think I'd like to try to keep them 64bits if possible. From reading the errata, it seems like moving these two fields https://github.com/gopcua/opcua/blob/master/monitor/subscription.go#L54-L55 to the "top" of the struct might fix it.

goodspeed1986, 2020-04-15:
Thanks, it works. I made fork of subscription.go. Thank you very much.

closed 2020-04-15 by goodspeed1986