gopcua/opcua #342: [Browse] The view timestamp is not available or not supported. StatusBadViewTimestampInvalid (0x80C90000)

felix-lessoer, 2020-04-17:
One of the users of my app based on this library is facing issues while browsing.
I found the following error in the logs:

```The view timestamp is not available or not supported. StatusBadViewTimestampInvalid (0x80C90000)```
This is happening while calling: `node.Children(refs, ua.NodeClassAll) ` where node is `*opcua.Node`

I think this is happening here: 
https://github.com/gopcua/opcua/blob/ea8d8232b87f654115782ae789a907e741e6f6e8/node.go#L185-L197

Any ideas how to solve that?

Original issue
https://github.com/elastic/Machinebeat/issues/12

cacamille3, 2020-05-07:
Hi,
I am from LOYTEC Electronics GmbH and responsible for the OPC UA Server on our devices.
We have received the issue from our customer using MachineBeat.
I guess the issue could be solved onthe OPC UA Client side by removing the Timestamp value in the ViewDescription.

ViewDescription's timestamp (or viewVersion) value shall ONLY be present when you want to browse a precise View of an OPC UA Address Space.
Since you do not set a ViewDescription's viewId value (or set a 0 value), it does not make sense to set the timestamp value.

cacamille3, 2020-05-07:
Hi,
ViewDescription's timestamp (or viewVersion) value shall ONLY be present when you want to browse a precise View of an OPC UA Address Space.
Since you do not set a ViewDescription's viewId value (or set a 0 value), it does not make sense to set the timestamp value.

felix-lessoer, 2020-05-07:
@magiconair 

I can create a PR that solves the problem, but want to hear your opinion which way to go:

1.) As long as we always set ViewID to 0 we can remove the timestamp completly.
OR
2.) We could make the ViewID a configurable item. If this is set we also set timestamp. If not we set it to 0 and do not set the timestamp.

Whats your opinion?

magiconair, 2020-05-07:
The more I look at that function the more I think that we need a different approach since there are so many configurable values. What we need is the `BrowseDescription` and some values for the `BrowseRequest` and we should have some helper functions to provide defaults. Maybe something along the lines of this:

```go
func (n *Node) References(refType uint32, dir ua.BrowseDirection, mask ua.NodeClass, includeSubtypes bool) ([]*ua.ReferenceDescription, error) {
	view := &ua.ViewDescription{
		ViewID:    ua.NewTwoByteNodeID(0),
		Timestamp: time.Now(),
	}
	return n.Browse(BrowseReferences(n.ID, refType, dir, mask, includeSubtypes), 0, view)
}

func BrowseReferences(id *ua.NodeID, refType uint32, dir ua.BrowseDirection, mask ua.NodeClass, includeSubtypes bool) *ua.BrowseDescription {
	if refType == 0 {
		refType = id.References
	}
	if mask == 0 {
		mask = ua.NodeClassAll
	}

	return &ua.BrowseDescription{
		NodeID:          id,
		BrowseDirection: dir,
		ReferenceTypeID: ua.NewNumericNodeID(0, refType),
		IncludeSubtypes: includeSubtypes,
		NodeClassMask:   uint32(mask),
		ResultMask:      uint32(ua.BrowseResultMaskAll),
	}
}

func (n *Node) Browse(desc *ua.BrowseDescription, maxRefs uint32, view *ua.ViewDescription) ([]*ua.ReferenceDescription, error) {
	req := &ua.BrowseRequest{
		View:                          view,
		RequestedMaxReferencesPerNode: maxRefs,
		NodesToBrowse:                 []*ua.BrowseDescription{desc},
	}

	resp, err := n.c.Browse(req)
	if err != nil {
		return nil, err
	}
	return n.browseNext(resp.Results)
}
```

Not happy with the `BrowseReferences` name yet. Maybe you have a better idea. But this would allow us to build common helpers for generating `BrowseDescription` objects for common browse requests without adding lots of arguments or adding another type.

What do you think?

magiconair, 2020-10-03:
We got bitten by that as well. I've dropped the `time.Now()`. Sorry, for the long delay.

Fixed by #388

closed 2020-10-03 by magiconair