gopcua/opcua #343: host godoc does not contain documentation for the new monitor package

jskelcy, 2020-04-25:
I imagine it is as simple as regenerating the documentation assets and pushing them to wherever the godoc is hosted, but I am not sure of the details. The new `monitor` package seems great and it would be nice to have all the documentation in one place.

magiconair, 2020-04-27:
Which new `monitor` package are you referring to? Is that still on a branch/PR? 

This one?

https://godoc.org/github.com/gopcua/opcua/monitor

jskelcy, 2020-04-27:
Wow, sorry about this. I did not realize how godoc lays out sub-packages, I was expecting to see all of the exposed types on the top level. While I think this would be nice for quickly searching the documentation, that is more of an issue with godoc then gopcua. 

Thanks for the quick response and I will close this out.

closed 2020-04-27 by jskelcy