gopcua/opcua #344: Question regarding status code Bad_NodeIdUnknown

jskelcy, 2020-04-25:
Hello team, 

New to the OPCUA space and I am using this library to pull data from an OPCUA server, I am currently testing against the Prosys OPCUA simulation server which seems to be the one a number of other people are using based on reading through comments and tickets.

The issue I am having is when I use a `NodeMonitor` (but I have observed the same issue with a `Subscription`) and I delete the Node I am monitoring I do not see any notification. From what I can tell from [the OPCUA reference](https://reference.opcfoundation.org/v104/Core/docs/Part4/5.12.1/) I should be seeing a `Bad_NodeIDUnknown` notification, here is the section of the reference I am looking at (bottom of the page):

> Deletion of a MonitoredItem should not be confused with the removal of the Attribute that it monitors. If the Node that contains the Attribute being monitored is deleted, the MonitoredItem generates a Notification with a StatusCode Bad_NodeIdUnknown that indicates the deletion, but the MonitoredItem is not deleted.

While this may be a failing of the simulator and less so of the library I am wondering if anyone else has run into this problem. After adding some debug statements and doing some testing I think it might be an issue with the simulator, but I am not sure. If I have misinterpreted the deletion of monitored nodes and there is a better way to do this please let me know.

Thank you so much, really enjoying this project so far.

magiconair, 2020-04-27:
Hi @jskelcy, I'm glad you like the project and that it works for you. I am not sure if we have tested deleting a monitored node and this could well be a bug. At least I haven't but in our setup things don't move that fast on the PLC side. However, that should be easy to test. Is that a blocker for you?

jskelcy, 2020-04-27:
I have a use case where I need to capture these events, but I am working on building out a test environment against a real OPUCA server, once I have something up and running I can update the ticket with the results.

jskelcy, 2020-04-28:
Did a bit of testing on a Kepware KEPServerEX that I have access to and got some interesting results. I was monitoring a node and deleted it and I got a `StatusBadConfigurationError` error, based on the reference I expected a `StatusBadNodeIDUnknown`, but that is not what Kepware is sending. 

In any event I think dont think there are any issues with the library as it did not swallow a non-data change event on a monitored node, so that's good for me. I am going to close this one out for now.

closed 2020-04-28 by jskelcy