gopcua/opcua #351: losing data due to the delay

yymemo, 2020-06-30:
> Spec 1.03 part 4 5.13.5 Publish
> [..] in high latency networks, the Client may wish to pipeline Publish requests to ensure cyclic reporting from the Server. Pipe-lining involves sending more than one Publish request for each Subscription before receiving a response. For example, if the network introduces a delay between the Client and the Server of 5 seconds and the publishing interval for a Subscription is one second, then the Client will have to issue Publish requests every second instead of waiting for a response to be received before sending the next request.
> 
Hi everyone, I have encountered the same question about losing data due to the delay. 
Are there any plans to implement the feature?
Thank you.

InasaXia, 2020-09-03:
same problem

magiconair, 2021-09-25:
Having more than one `PublishRequest` in flight can be useful. I'll have a look how hard it is to add this.

magiconair, 2021-09-25:
The v0.1.x implementation produced one `PublishRequest` per subscription which leads to problems on our end if we have too many subscriptions probably b/c we don't support client chunking which would allow us to subscribe to a large number of nodes. The v0.2.x implementation produces one `PublishRequest` period. That is the other extreme. In low-latency networks like ours this is probably not as much of an issue.

At a first glance there isn't a straightforward way to add this since the pause/resume/error logic depends on this. I can see how to keep a number of go-routines running in the background.