gopcua/opcua #352: Subscribe.Monitor response, StatusCode is not correct.

yymemo, 2020-06-30:
**opcua server-node information**
Ex:StatusBadWaitingForInitialData(0x80320000) or StatusUncertain(0x40000000)

**Subscribe.Monitor response code**
`resp, err := sub.Monitor(ua.TimestampsToReturnBoth, toAdd...)
	if err != nil || resp.Results[0].StatusCode != ua.StatusOK {
		return err
	}
`
I check the resp.Results[0].StatusCode=0, why I can't get actual status code(Ex:0x80320000 or other value)?

magiconair, 2020-07-24:
Could this be related to #337 ?

magiconair, 2020-10-03:
Since we are refactoring the subscription logic for the next release I would like to ask you to test the release candidate when we have it. Currently, the code in #385 is the most relevant.

magiconair, 2021-11-11:
@ymmemo unless I'm missing something we are not modifying the response we are getting from the server. You should get the unmodified response from the server or the first status code that is not `StatusOK` in the `error`. 

Which server are you testing this against? Can you provide a tcpdump?

closed 2025-04-28 by github-actions[bot]