gopcua/opcua #356: Subscribe to multiple nodeID at examples/subscribe.go

78planet, 2020-07-07:
Can I subscribe to multiple nodeID at examples/subscribe.go?
I want to subscribe multiple nodes in one file.

78planet, 2020-07-07:
It is possible to receive multiple nodeID as shown below.
`go run subscribe.go -endpoint opc.tcp://localhost:4840 -node 'ns=1;s=Counter' -node 'ns=1;s=Sawtooth`

```
type arrayFlags []*string

func (i *arrayFlags) String() string {
	return "my string representation"
}

func (i *arrayFlags) Set(value string) error {
	*i = append(*i, &value)
	return nil
}

var nodeIDarr arrayFlags

flag.Var(&nodeIDarr, "node", "node id to subscribe to")

```

The code below is wrapped in a for statement.
```	
var miCreateRequest *ua.MonitoredItemCreateRequest
var eventFieldNames []string
for _, nodeID := range nodeIDarr {
	id, err := ua.ParseNodeID(*nodeID)
	if err != nil {
		log.Fatal(err)
	}

	if *event {
		miCreateRequest, eventFieldNames = eventRequest(id)
	} else {
		miCreateRequest = valueRequest(id)
	}
	res, err := sub.Monitor(ua.TimestampsToReturnBoth, miCreateRequest)
	if err != nil || res.Results[0].StatusCode != ua.StatusOK {
		log.Fatal(err)
	}
	go sub.Run(ctx) // start Publish loop
}
```
Solved easily.

78planet, 2020-07-07:
Using the above code, you can print data from multiple nodes.
Data of multiple nodes is output. But I don't know the data of a specific node.
Is there a way to know the data of a specific node from multiple data?

kung-foo, 2020-07-24:
Hi @78planet, you might want to look at the higher level monitor interface: https://pkg.go.dev/github.com/gopcua/opcua@v0.1.11/monitor?tab=doc and https://github.com/gopcua/opcua/tree/master/examples/monitor

This abstracts away some of the complexity of the lower level subscription requirements (e.g. manually providing distinct "handles" for each subscribed item).

78planet, 2020-07-24:
Thank you @kung-foo, It was very helpful.

closed 2020-07-24 by 78planet