gopcua/opcua #362: ua/variant: index out of range

ceh, 2020-08-15:
Fuzzed the implementation of `ua.DecodeService` with go-fuzz, and got the following runtime error:

```
panic: runtime error: index out of range [0] with length 0

goroutine 1 [running]:
github.com/gopcua/opcua/ua.split(0x0, 0x0, 0x0, 0xc000015480, 0x2, 0x2, 0x502660, 0xc00000cc60, 0x97, 0x1, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/variant.go:215 +0x7a7
github.com/gopcua/opcua/ua.(*Variant).Decode(0xc000107380, 0x7f7fe03c0038, 0x18, 0x18, 0xc000107380, 0xc000107380, 0x64e860)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/variant.go:187 +0x6f5
github.com/gopcua/opcua/ua.decode(0x7f7fe03c0038, 0x18, 0x18, 0x5332a0, 0xc000052440, 0x196, 0xc000018500, 0x20, 0x0, 0x0, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:50 +0x228
github.com/gopcua/opcua/ua.decodeStruct(0x7f7fe03c002a, 0x26, 0x26, 0x52eba0, 0xc000052420, 0x199, 0x4f7016, 0x1a, 0x0, 0x1c400000002, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:108 +0x361
github.com/gopcua/opcua/ua.decode(0x7f7fe03c002a, 0x26, 0x26, 0x52eba0, 0xc000052420, 0x199, 0x4f7016, 0x1a, 0x0, 0x0, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:85 +0xd49
github.com/gopcua/opcua/ua.decode(0x7f7fe03c002a, 0x26, 0x26, 0x4fce80, 0xc000052420, 0x16, 0x4f7016, 0x1a, 0x0, 0x0, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:83 +0xc45
github.com/gopcua/opcua/ua.Decode(0x7f7fe03c002a, 0x26, 0x26, 0x4fce80, 0xc000052420, 0x0, 0xc00005e2a0, 0x54a008)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:35 +0xfc
github.com/gopcua/opcua/ua.(*Buffer).ReadStruct(0xc0001452f8, 0x4fce80, 0xc000052420)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/buffer.go:172 +0xf2
github.com/gopcua/opcua/ua.(*ExtensionObject).Decode(0xc00000cbe0, 0x7f7fe03c001e, 0x32, 0x32, 0xc00000cbe0, 0xc00000cbe0, 0x64e860)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/extension_object.go:82 +0x4b8
github.com/gopcua/opcua/ua.decode(0x7f7fe03c001e, 0x32, 0x32, 0x511e60, 0xc00005c720, 0x196, 0xc00001c280, 0x3c, 0x0, 0x0, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:50 +0x228
github.com/gopcua/opcua/ua.decodeStruct(0x7f7fe03c0004, 0x4c, 0x4c, 0x52b2e0, 0xc00005c6e0, 0x199, 0xc00001a210, 0x2b, 0x8000000002, 0x0, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:108 +0x361
github.com/gopcua/opcua/ua.decode(0x7f7fe03c0004, 0x4c, 0x4c, 0x52b2e0, 0xc00005c6e0, 0x199, 0xc00001a210, 0x2b, 0x0, 0x0, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:85 +0xd49
github.com/gopcua/opcua/ua.decode(0x7f7fe03c0004, 0x4c, 0x4c, 0x50bca0, 0xc00000e0a8, 0x196, 0xc00001a210, 0x2b, 0x0, 0x0, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:83 +0xc45
github.com/gopcua/opcua/ua.decodeStruct(0x7f7fe03c0004, 0x4c, 0x4c, 0x5129e0, 0xc00000e0a8, 0x199, 0x4f79b5, 0x1d, 0x0, 0x0, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:108 +0x361
github.com/gopcua/opcua/ua.decode(0x7f7fe03c0004, 0x4c, 0x4c, 0x5129e0, 0xc00000e0a8, 0x199, 0x4f79b5, 0x1d, 0x0, 0x0, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:85 +0xd49
github.com/gopcua/opcua/ua.decode(0x7f7fe03c0004, 0x4c, 0x4c, 0x50e760, 0xc00000e0a8, 0x16, 0x4f79b5, 0x1d, 0x0, 0x0, ...)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:83 +0xc45
github.com/gopcua/opcua/ua.Decode(0x7f7fe03c0004, 0x4c, 0x4c, 0x50e760, 0xc00000e0a8, 0x0, 0x0, 0x9)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/decode.go:35 +0xfc
github.com/gopcua/opcua/ua.DecodeService(0x7f7fe03c0004, 0x4c, 0x50, 0xc000145e98, 0x46d406, 0x5f314682, 0x12443b42, 0x100ae1f304fff)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/service.go:51 +0x285
github.com/gopcua/opcua/ua.Fuzz(0x7f7fe03c0000, 0x50, 0x50, 0x4)
	/home/ceh/code/src/github.com/gopcua/opcua/ua/fuzz.go:6 +0x5b
go-fuzz-dep.Main(0xc000145f70, 0x1, 0x1)
	go-fuzz-dep/main.go:36 +0x1ad
main.main()
	github.com/gopcua/opcua/ua/go.fuzz.main/main.go:15 +0x52
exit status 2
```

To reproduce:
```
package main

import "github.com/gopcua/opcua/ua"

func main() {
	data := "\x010\xc4\x010000000000000000" +
		"00\xff\xff\xff\xff0000\x12\x00\x00n\x01\x00\x000&\x00" +
		"\x00\x0000000000000000\xc9\x00\x00\x00" +
		"\x00\x02\x00\x00\x00\x00\x00\x00\x0000000000000"

	ua.DecodeService([]byte(data))
}

```

closed 2020-08-16 by magiconair

magiconair, 2020-08-16:
Cool that you've fuzzed it. Wanted to do this for a while in a project. Did you find anything else?

Can we integrate this into the CI flow somehow?

ceh, 2020-08-16:
> Can we integrate this into the CI flow somehow?

There's [OSS-Fuzz](https://google.github.io/oss-fuzz/), which [open62541](https://github.com/open62541/open62541/tree/master/tests/fuzz) uses. It's available for free to open source projects.

See also https://google.github.io/oss-fuzz/getting-started/accepting-new-projects/ and https://google.github.io/oss-fuzz/getting-started/new-project-guide/go-lang/

I suppose it's also possible to setup something custom using GitHub Actions if OSS-Fuzz doesn't accept the project.