gopcua/opcua #366: Unable to connect to ProSys Simulation Server v5

haylesnortal, 2020-08-23:
The latest version of ProSys Simulation Server (5.0.0-217) is returning the following error when trying to connect:

`Error in Client Connection: An internal error occurred as a result of a programming or configuration error. StatusBadInternalError (0x80020000)`

I suspect it's an issue with their new version but want to ask here for confirmation. The ProSys Forum had one report of a similar issue with the new version and the advice was that there's an issue with v5's handling of Application Name.
[link](https://forum.prosysopc.com/forum/opc-ua-simulation-server/issue-while-connecting-new-version-of-prosys-5-0-0-217-due-to-latest-versions-of-bcp-jars/)

I've attempted to make sure I am configuring as much of the identity properties as possible and the result is the same.

My testing code:
`func main() {

	endpoint := "opc.tcp://10.0.0.244:53530/OPCUA/SimulationServer"
	ctx := context.Background()
	mode := "None"
	policy := "None"

	endpoints, err := opcua.GetEndpoints(endpoint)
	if err != nil {
		log.Fatal(err)
	}
	ep := opcua.SelectEndpoint(endpoints, policy, ua.MessageSecurityModeFromString(mode))
	if ep == nil {
		log.Fatal("Failed to find suitable endpoint")
	}

	opts := []opcua.Option{
		opcua.SecurityPolicy(policy),
		opcua.SecurityModeString(mode),
		opcua.AuthAnonymous(),
		opcua.SecurityFromEndpoint(ep, ua.UserTokenTypeAnonymous),
		opcua.ApplicationURI("urn:gopcua:client"),
		opcua.ApplicationName("myapp"),
		opcua.ProductURI("product-uri"),
		opcua.RequestTimeout(time.Second * 30),
	}

	o := opcua.NewClient(endpoint, opts...)
	if err = o.Connect(ctx); err != nil {
		log.Printf("Error in Client Connection: %s", err)
	}
}
`

On the ProSys side I can see sessions attempting to connect but they're showing no identity information and the timeout property doesn't look to be getting updated either.

[image]

Any chance this is on the gopcua end? Seems unlikely to me. Or am I doing something wrong - another likely candidate?

magiconair, 2020-08-24:
On first glance this looks ok. Any logging on the Prosys side?

kung-foo, 2020-08-24:
heh, Prosys is sending back a Java stacktrace:
[image]

```
Eþ×³@@bDÑVdú%ï£Pó
Èô5Èô5MSGFÊônsport.tcp.nio.g.d(SourceFile:862)
	at com.prosysopc.ua.stack.transport.tcp.nio.g$3.e(SourceFile:263)
	at com.prosysopc.ua.stack.transport.tcp.nio.j.bwi(SourceFile:442)
	at com.prosysopc.ua.stack.transport.tcp.nio.j.c(SourceFile:493)
	at com.prosysopc.ua.stack.transport.tcp.nio.j$1.run(SourceFile:146)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)

", StringTable="[java.lang.NullPointerException: clientIdentity]", AdditionalHeader="null"]"].java.lang.NullPointerException: clientIdentity
```

kung-foo, 2020-08-24:
We are not creating the `LocalizedText` correctly for the `ApplicationName`.

```diff
diff --git a/config.go b/config.go
index e62a644..5269156 100644
--- a/config.go
+++ b/config.go
@@ -64,7 +64,7 @@ type Option func(*uasc.Config, *uasc.SessionConfig)
 // ApplicationName sets the application name in the session configuration.
 func ApplicationName(s string) Option {
        return func(c *uasc.Config, sc *uasc.SessionConfig) {
-               sc.ClientDescription.ApplicationName = &ua.LocalizedText{Text: s}
+               sc.ClientDescription.ApplicationName = &ua.LocalizedText{Text: s, EncodingMask: ua.LocalizedTextText}
        }
 }
```

PR #367 (2020-08-24): Add localized text helpers
Fixes #366
closed 2020-08-24 by magiconair