gopcua/opcua #375: [Subscribe] question about handle

ymmemo, 2020-09-04:
Hello
I am trying to subscribe to more nodes.
But I have a question about examples/subscribe/subscribe.go
`handle:= uint32(42)`
what does 42 mean? Can I assign another number?

unknownet, 2020-09-05:
To be confirmed, but from what I understand the `handle` is an identifier that the client use to mark a `monitoredItem`.
The server send multiple unsorted `DataChangeNotification` and the only way to know from wich `monitoredItem` the  `DataChangeNotification` came from, is to compare their `handle`.

Note: the `handle` has to be unique (global) or you are going to mix up the notification, also the `handle` mechanism works well with a map (`map[uint32]*ua.NodeID` for instance) to identify the related NodeID.

You don't necessary have to know what is an handle if you use the API ["github.com/gopcua/opcua/monitor"]() instead, it contains an abstraction layer that take care of `handle` for you under the hood.

ymmemo, 2020-09-07:
@unknownet I see. thank you for your help : )

closed 2020-10-03 by magiconair