gopcua/opcua #376: Data race in secure channel and

antlad, 2020-09-04:
Valid for 0.1.11 and 0.1.12 (didn't tried older). 
Tested with server https://github.com/Azure-Samples/iot-edge-opc-plc . 
Easy way to run `docker run --rm --network=host --mount type=bind,source="$(pwd)"/pki,target=/appdata/pki mcr.microsoft.com/iotedge/opc-plc:latest` . It will mount pki folder to local, so you can your certs on first connect from rejected to trusted. 
Looks like it happen on session timeout. Initially everything is working. Then I see this stacktrace (with -race flag). 
Next I see line `invalid response: got <nil> want **ua.PublishResponse` in `Monitor` `SetErrorHandler`.
And then no reconnection attempts and no new monitor data callbacks.

`==================
WARNING: DATA RACE
Write at 0x00c0003042e8 by goroutine 28:
  xxx/vendor/github.com/gopcua/opcua/uasc.(*SecureChannel).openSecureChannel()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/uasc/secure_channel.go:701 +0x157
  xxx/vendor/github.com/gopcua/opcua/uasc.(*SecureChannel).Renew-fm()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/uasc/secure_channel.go:660 +0x49
  xxx/vendor/github.com/gopcua/opcua.(*Client).openSecureChannel()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/client.go:193 +0x38
  xxx/vendor/github.com/gopcua/opcua.(*Client).scheduleRenewingToken.func1()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/client.go:773 +0x1b0

Previous read at 0x00c0003042e8 by goroutine 51:
  xxx/vendor/github.com/gopcua/opcua/uasc.(*SecureChannel).signAndEncrypt()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/uasc/secure_channel_crypto.go:35 +0x1db
  xxx/vendor/github.com/gopcua/opcua/uasc.(*SecureChannel).sendAsyncWithTimeout()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/uasc/secure_channel.go:210 +0x20a
  xxx/vendor/github.com/gopcua/opcua/uasc.(*SecureChannel).SendRequestWithTimeout()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/uasc/secure_channel.go:161 +0xa2
  xxx/vendor/github.com/gopcua/opcua.(*Client).sendWithTimeout()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/client.go:493 +0xf5
  xxx/vendor/github.com/gopcua/opcua.(*Subscription).publish()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/subscription.go:143 +0x248
  xxx/vendor/github.com/gopcua/opcua.(*Subscription).Run()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/subscription.go:174 +0xc6

Goroutine 28 (running) created at:
  xxx/vendor/github.com/gopcua/opcua.(*Client).scheduleRenewingToken()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/client.go:766 +0xf3
  xxx/vendor/github.com/gopcua/opcua.(*Client).openSecureChannel()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/client.go:199 +0x120
  xxx/vendor/github.com/gopcua/opcua.(*Client).Dial()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/client.go:189 +0x3b0
  xxx/vendor/github.com/gopcua/opcua.(*Client).Connect()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/client.go:146 +0xb5
  xxx/drivers/opcua.(*opcuaClient).Init()
      ${GOPATH}/xxx/drivers/opcua/opcua.go:754 +0x16c
  xxx/handler/device.(*device).Run()
      ${GOPATH}/xxx/handler/device/device.go:510 +0x862

Goroutine 51 (running) created at:
  xxx/vendor/github.com/gopcua/opcua/monitor.(*NodeMonitor).Subscribe()
      ${GOPATH}/xxx/vendor/github.com/gopcua/opcua/monitor/subscription.go:131 +0x17f
  xxx/drivers/opcua.(*opcuaClient).updateSubs()
      ${GOPATH}/xxx/drivers/opcua/opcua.go:688 +0x5e9
  xxx/drivers/opcua.(*opcuaClient).Init()
      ${GOPATH}/xxx/drivers/opcua/opcua.go:770 +0x49f
  xxx/handler/device.(*device).Run()
      ${GOPATH}/xxx/handler/device/device.go:510 +0x862
`

antlad, 2020-09-04:
I can provide environment on request if that help to solve issue

magiconair, 2021-11-09:
We removed all the races in the secure channel implementation.

closed 2021-11-09 by magiconair