gopcua/opcua #384: Bounty offer for looking into #368 and #376

dtoubelis, 2020-09-15:
May we offer you a bounty for resolving #368 and #376?

magiconair, 2020-09-15:
Hi Dmitri, thank you for reaching out. 

I cannot speak for the other contributors but I’d be willing to throw in some extra hours if you make a donation to something that makes the world a better place. My list includes in no particular order

* GoBridge - https://github.com/gobridge/about-us/blob/master/README.md
* Fridays for Future
* The International Red Cross
* Médecins Sans Frontières
* Greenpeace 

Or something else that you think is worthwhile - maybe something more local to you. 

To other contributors: please feel free provide your opinion as well. 

Please note that we will fix them eventually but I’m quite busy with hiring, day job and personal things. If you’re OK with making a donation I will throw in some evenings and night shifts to push this forward. 

So the most helpful thing you can do is to test the changes and provide feedback on the issue so that we see who needs this fix. 

But again: thank you for reaching out and welcome to the community ❤️
—
Frank Schröder

dtoubelis, 2020-09-15:
Thanks. I've sent $200 to the first one on the list.

dtoubelis, 2020-09-21:
Hi, is there any progress?

magiconair, 2020-09-22:
Not yet. Injured my foot last week and got high workload at day job.

magiconair, 2020-10-02:
@dtoubelis can you have a look at #368 and start testing that? It does not address the race condition yet but I need to check whether that is still relevant after the refactoring.

magiconair, 2021-11-16:
Hi @dtoubelis can we consider this closed?

magiconair, 2021-11-16:
Maybe try with v0.2.1 which I've just released.

dtoubelis, 2021-11-16:
We are no longer detecting any race conditions. so looks like it is resolved. Thank you for your help.

closed 2021-11-16 by dtoubelis