gopcua/opcua #392: scheduleRenewal not start when security mode is None

ahuret, 2020-10-21:
Hi all !

We're implementing a prometheus exporter that scrap metrics from an opcua server and we faced an interesting issue:

For test purpose, we're using Anonymous authentication mode and None security mode.
Every ~ 1 hour, we had issue with connection. We found that this 1 hour correlate with default SecureChannel Lifetime setting.
But we'd still to get how to refresh connection until it's expired. We tested multiple stuff, custom renew implementation first (doesn't works), then we decided to use 298-reconnect-reactivate branch, but still no reconnection.
We finally found https://github.com/gopcua/opcua/blob/298-reconnect-reactivate/uasc/secure_channel.go#L600 and scheduleRenewal that schedule the SecurityToken renewal and it seems this is exactly what we've looking for.

My question here is:

	if s.cfg.SecurityMode != ua.MessageSecurityModeNone {
		go s.scheduleRenewal(instance)

specifies that renewal only occurs when security mode is not None, but seems in our case we need it, why that decision then ?

Thanks a lot

jbguerraz, 2020-10-21:
Maybe @kung-foo would you have a clue ? Thank you a lot!

jbguerraz, 2020-11-02:
Can't find evidence of such condition in https://reference.opcfoundation.org/v104/Core/docs/Part4/5.5.2/

VereeckenWannes, 2021-09-16:
Hi!
We're also having this issue. Using v0.1.13 this issue is the most visible in the form of a lost connection.
Later versions seem to hide the problem when using the AutoReconnect option.
Removing the if statement mentioned above seems to fix the problem for us, so I did some digging in the OPC-UA documentation and this was what I found regarding MessageSecurityModeNone.

https://reference.opcfoundation.org/v104/Core/docs/Part4/5.5.2/#Table11
> A SecureChannel may have to be created even if the securityMode is NONE. The exact behaviour depends on the mapping used and is described in the OPC 10000-6.

https://reference.opcfoundation.org/v104/Core/docs/Part6/6.7.4/#:~:text=See%20OPC%2010000,or%20expired%20TokenIds.
> See OPC 10000-2 for more information on the requirements for random number generators. **The OpenSecureChannel Messages are not signed or encrypted if the SecurityMode is None**. The Nonces are ignored and should be set to null. The SecureChannelId and the TokenId are still assigned but no security is applied to Messages exchanged via the channel. **The SecurityToken shall still be renewed before the RevisedLifetime expires**. Receivers shall still ignore invalid or expired TokenIds

magiconair, 2021-09-23:
Fixed by #466

closed 2021-09-23 by magiconair