gopcua/opcua #393: Node returns incorrect Namespace

fr-ser, 2020-10-24:
Hi,
sadly I am only a user of an OPC/UA server and neither an expert nor its administrator. But I had the following problem using a CLI tool I built to read data from the OPC/UA server regarding the.

When reading the namespace from the node I receive `1` even though `4` is the correct namespace.
The code that returns `1` is the following
```go
attrs, err := node.Attributes(ua.AttributeIDBrowseName, ua.AttributeIDValue, ua.AttributeIDNodeID)
# some other stuff between
treeNode.ns = int16(attrs[2].Value.NodeID().Namespace())
```

But I definitely have to specify `4` to read a value. Below you can find the output of the CLI that demonstrates the issue.
Below I generate a tree of the OPC/UA nodes and I get back a `1:PLC1`, whereas `4:PLC1` would be correct...
As far as I tested, this is the only node with an incorrect namespace

```txt
> go-opcua-cli.exe list -d 2

0:Objects
├─ 0:Server
│  ├─ 0:ServerArray ([urn:CX-3B1220:BeckhoffAutomation:TcOpcUaServer:1])
│  ├─ 0:NamespaceArray ([http://opcfoundation.org/UA/ urn:CX-3B1220:BeckhoffAutomation:TcOpcUaServer:1 http://opcfoundation.org/UA/DI/ http://PLCopen.org/OpcUa/IEC61131-3/ urn:CX-3B1220:BeckhoffAutomation:Ua:PLC1 urn:BeckhoffAutomation:Ua:Types:GlobalTypes http://Beckhoff.com/TwinCAT/TF6100/Server/Configuration])
│  ├─ 0:ServerStatus (&{1 i=864 0xc0001aa4b0})
│  ├─ 0:ServiceLevel (255)
│  ├─ 0:Auditing (false)
│  ├─ 0:ServerCapabilities
│  ├─ 0:ServerDiagnostics
│  ├─ 0:VendorServerInfo
│  ├─ 0:ServerRedundancy
│  ├─ 0:GetMonitoredItems
│  └─ 1:Trace
├─ 2:DeviceSet
│  └─ 1:DefaultConfiguration
└─ 1:PLC1
   ├─ 1:DeviceManual
   ├─ 1:DeviceRevision
    |─ 1:HardwareRevision
   ├─ 1:Manufacturer (&{3  Beckhoff})
   ├─ 1:Model (&{0  })
   ├─ 1:RevisionCounter (0)
   ├─ 1:SerialNumber
   ├─ 1:SoftwareRevision
   ├─ 1:Programs
   ├─ 1:Tasks
   ├─ 4:GVL
   ├─ 4:GVL_NMEA
   └─ 4:DeviceState

> go-opcua-cli.exe read -p 0:Objects/4:PLC1

NodeClass: Object
BrowseName: PLC1
Description: The attribute is not supported for the specified Node. StatusBadAt
ributeIDInvalid (0x80350000)
DataType:
Value:

> go-opcua-cli.exe read -p 0:Objects/1:PLC1

2020/10/24 13:58:02 Error parsing path '0:Objects/1:PLC1'
The requested operation has no match to return. StatusBadNoMatch (0x806F0000)

```

magiconair, 2021-11-09:
@fr-ser is this still an issue? Could you provide a tcpdump?

fr-ser, 2021-11-10:
Sorry, but I don't have access to that specific OPC UA server anymore and cannot test it anymore...

magiconair, 2021-11-11:
Do you know what OPC/UA server this was?

fr-ser, 2021-11-15:
Sadly not.

kung-foo, 2021-11-15:
The namespace array seems to indicate the server:

`[http://opcfoundation.org/UA/ urn:CX-3B1220:BeckhoffAutomation:TcOpcUaServer:1 http://opcfoundation.org/UA/DI/ http://PLCopen.org/OpcUa/IEC61131-3/ urn:CX-3B1220:BeckhoffAutomation:Ua:PLC1 urn:BeckhoffAutomation:Ua:Types:GlobalTypes http://Beckhoff.com/TwinCAT/TF6100/Server/Configuration]`

closed 2025-04-27 by github-actions[bot]