gopcua/opcua #394: Cannot retrieve node Attributes (invalid extension)

fr-ser, 2020-10-24:
Hi,
sadly I am only a user of an OPC/UA server and neither an expert nor its administrator. But I had the following problem using a CLI tool I built to read data from the OPC/UA server regarding the.

For reading the cli uses basically this code
```go
attrs, err := node.Attributes(
	ua.AttributeIDNodeClass,ua.AttributeIDBrowseName,
	ua.AttributeIDDescription,ua.AttributeIDDataType, ua.AttributeIDValue,
)
if err != nil {
	return fmt.Errorf("Error retrieving attributes of node %s\n%s", node.String(), err)
}
```

I could not get the attributes of a node and got the following error message
```txt
> go-opcua-cli.exe read -p 0:Objects/4:PLC1/4:GVL/4:IntService

2020/10/24 14:11:53 Error retrieving attributes of node ns=4;s=GVL.IntService
opcua: invalid extension object with id ns=4;s=<StructuredDataType>:tIntService__DefaultBinary
```

Using an OPC/UA [python library](https://github.com/FreeOpcUa/opcua-asyncio) I go the following result for the same command
```txt
> opcua-cli.exe read -p 0:Objects/4:PLC1/4:GVL/4:IntService

BrowseName: 4:IntService
DisplayName: LocalizedText(Encoding:3, Locale:, Text:IntService)
Description:
NodeClass: Variable

Value: [ExtensionObject(TypeId:ns=4;s=<StructuredDataType>:tIntService__DefaultB
inary, Encoding:1, 47 bytes), ExtensionObject(TypeId:ns=4;s=<StructuredDataType>
:tIntService__DefaultBinary, Encoding:1, 47 bytes), ExtensionObject(TypeId:ns=4;
s=<StructuredDataType>:tIntService__DefaultBinary, Encoding:1, 47 bytes), Extens
ionObject(TypeId:ns=4;s=<StructuredDataType>:tIntService__DefaultBinary, Encodin
g:1, 47 bytes), ExtensionObject(TypeId:ns=4;s=<StructuredDataType>:tIntService__
DefaultBinary, Encoding:1, 47 bytes)]
```

magiconair, 2020-10-25:
Thank you for reaching out. 

The gopcua library does not handle extension objects dynamically. That means you have to register them before you can use them. Check the examples. There should be one where a custom extension object is registered.

fr-ser, 2020-10-25:
Thank you for your answer. I found the example here (for future reference): 
https://github.com/gopcua/opcua/blob/master/examples/udt/udt.go#L92

I will read up on extension objects in OPC UA and see how I can deal with them 👌

closed 2020-10-25 by fr-ser