gopcua/opcua #395: 3K nodes monitoring - timeouts/broken pipe issues

alexlev2004, 2020-10-26:
Hi,

I have a setup with 3K nodes monitored every 1 sec. It is working well with open62541 based client (C language library). However, when I am trying to use gopcua, I am experiencing the below:
1) When trying to subscribe in single monitor,  then I see a timeout
2) splitting in different monitors each handling 50-500 nodes will be working, but from time to time experience a "broken pipe" error reported requiring to reestablish the connection.

Q1:
Do you have any best practices to apply in this case?    

Q2:
What could be probably wrong with my implementation?

Thanks
Alex

alexlev2004, 2020-10-29:
Just observed the Error: Bad Sequence returned by OPC server...
And indeed it can be seen from debug messages that 6392 comes after 6393:
 
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: uasc 1/6387: recv MSGF with 7982 bytes
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: uasc 1/6391: send *ua.PublishRequest with 74 bytes
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: uasc 1/6387: recv *ua.PublishResponse
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: sending *ua.PublishResponse to handler
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: conn 1: recv MSGF with 7220 bytes
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: uasc 1/6382: recv MSGF with 7220 bytes
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: uasc 1/6382: recv *ua.PublishResponse
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: sending *ua.PublishResponse to handler
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: conn 1: recv MSGF with 7592 bytes
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: uasc 1/6385: recv MSGF with 7592 bytes
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: uasc 1/6393: send *ua.PublishRequest with 74 bytes
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: uasc 1/6385: recv *ua.PublishResponse
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: sending *ua.PublishResponse to handler
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: uasc 1/6394: send *ua.PublishRequest with 74 bytes
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: conn 1: recv ERRF with 49 bytes
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: uasc readChunk EOF
Oct 28 16:07:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: uasc 1/6392: send *ua.PublishRequest with 74 bytes
Oct 28 16:08:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: debug: subscription 223 Run loop stopped
Oct 28 16:08:47 halo-OptiPlex-7040 halodigital/opcua-client:0.45/opcua-client-0.45/86ecd1b465c7[2716]: opcua-client@14:08:47.416096 ERROR|main.go:162|error: sub=225 err=write tcp 10.61.61.29:37586->10.61.61.194:49320: write: connection reset by peer
h

magiconair, 2020-11-08:
Which version are you using?

alexlev2004, 2020-11-08:
master

diericd, 2021-09-17:
Hi,

I can confirm this issue. It due to the fact that chunking is not implemented so any message with a size bigger than your tcp-window size will not be handled correctly. 

We also have to minimize the number of nodes we can read out each request but this greatly reduces the number of nodes we can read per seconds. 

Are there any plans to implement message chunking in the near future?

Thanks,

David

kung-foo, 2021-09-17:
Message chunking _is_ implemented in server -> client communication. I assume you mean client -> server chunking? If so, then yes, it appears to not be implemented.

TCP window size shouldn't come into play as that is farther down the networking stack than OPC cares about (unless I am misunderstanding the issue).

At Intelecy we sorta get around this issue by creating a single monitor, and then updating the subscription list in chunks. I suppose we'll run into the lack of client to server chunking if and when the monitored items notification ack is too big.

diericd, 2021-09-21:
I'm indeed talking about client -> server chunking. I had indeed forgotten where the maximum message size was coming from, I now remember it is negotiated at time of connection. 

We also try to get around this by splitting up the requests but we run into performance issues when trying to polling a large number of nodes in a certain time frame. 
It takes much longer to process 5 request for 100 nodes than 1 query for 500 nodes and this limits us in the number of values we can request per second.

magiconair, 2021-09-22:
This sounds like something we might also are going to run into but I'd like to get v0.2 over the line first. Lets schedule this for v0.3

magiconair, 2021-09-22:
We do in fact split the monitored items in multiple subscriptions to get around this. So adding client -> server chunking might simplify our code as well.

diericd, 2021-09-29:
Hi, 

I've made a branch in which I implemented client-chunking ( https://github.com/diericd/opcua/tree/client-chuncking ).  I've tested it on a few different OPC-UA servers  with plaintext and encryption and it seems to work. Should I make a pull request?

kung-foo, 2021-09-29:
Yes, that would be great!!!

closed 2025-04-27 by github-actions[bot]