gopcua/opcua #396: scheduleRenewal called only when SecurityMode is not MessageSecurityModeNone

alexlev2004, 2020-10-27:
Hi,

I noticed that my session is disconnected every hour. I guess it is because of the fact that scheduleRenewal called only when SecurityMode is not MessageSecurityModeNone in method handleOpenSecureChannelResponse.

Can you please elaborate why this condition is required? I think that secure channel need to be renewed even in case the Message Security Mode is "None". 

What do you think?

Thanks.
Alex

jbguerraz, 2020-11-10:
duplicates #392

alexlev2004, 2020-11-10:
Duplicates #392 + Fixed in the master

closed 2020-11-10 by alexlev2004

jbguerraz, 2020-11-10:
@alexlev2004  https://github.com/gopcua/opcua/blob/master/uasc/secure_channel.go#L600 looks like it's still there or am I missing something ?

alexlev2004, 2020-11-10:
> @alexlev2004 https://github.com/gopcua/opcua/blob/master/uasc/secure_channel.go#L600 looks like it's still there or am I missing something ?

Right, edited my comment right now
Thanks