gopcua/opcua #399: ERR: The size of the message specified in the header is too large

alexlev2004, 2020-11-08:
Subscribing for 3K nodes in single monitor causes an error

magiconair, 2020-11-08:
Which version are you using?

alexlev2004, 2020-11-08:
master

magiconair, 2020-11-09:
We have merged some significant changes to master regarding subscriptions. Is it possible for you to try them and see if the error persists?

magiconair, 2020-11-09:
FWIW, we are also not subscribing to all nodes at the same time. We are subscribing to 50 nodes at a time in a loop until all nodes are registered.

It is possible that depending on the node ids and the number of nodes you want to subscribe to you exceed the maximum size for a single message. 

Try to split this up in chunks of 50 or 100 nodes. This might also reduce the load on the OPC/UA server.

alexlev2004, 2020-11-09:
Yes if I split it works.
I tried to avoid concurrency and work around problems like #398 which seems having it's roots in thread safeness.

alexlev2004, 2020-11-09:
> We have merged some significant changes to master regarding subscriptions. Is it possible for you to try them and see if the error persists?

Sure. Testing it now and will update. Thanks for your support

alexlev2004, 2020-11-10:
Although can be done behind the scenes, we apply the workaround: split to separate monitors and subscribe for 50-100 items at once

closed 2020-11-10 by alexlev2004