gopcua/opcua #407: Problem to Get attribute Value

darlandieterich, 2020-12-02:
Always my Value is empty object '{}'. In my server I have setted value in the specified node. 
My question is, how get correct value?

```
attrs, err := node.Attributes(
		ua.AttributeIDNodeClass,
		ua.AttributeIDBrowseName,
		ua.AttributeIDDataType,
		ua.AttributeIDValue)

switch err := attrs[3].Status; err {
		case ua.StatusOK:
			def.Value = attrs[3].Value
		case ua.StatusBadAttributeIDInvalid:
			// ignore
		default:
			return NodeDef{}, err
	}
```
return recursive:

```
{
      "NodeID": "ns=4;i=141",
      "NodeClass": 2,
      "BrowseName": "EL1",
      "Description": "",
      "AccessLevel": 0,
      "Path": "Server interface_1.EL1",
      "DataType": "nada",
      "Value": null,
      "Childs": [
        {
          "NodeID": "ns=4;i=142",
          "NodeClass": 2,
          "BrowseName": "Nome",
          "Description": "",
          "AccessLevel": 0,
          "Path": "Server interface_1.EL1.Nome",
          "DataType": "nada",
          "Value": {},
          "Childs": []
        },
        {
          "NodeID": "ns=4;i=143",
          "NodeClass": 2,
          "BrowseName": "Liga",
          "Description": "",
          "AccessLevel": 0,
          "Path": "Server interface_1.EL1.Liga",
          "DataType": "bool",
          "Value": {},
          "Childs": []
        }
}
```

felix-lessoer, 2020-12-05:
You can have a look here:

https://github.com/elastic/Machinebeat/blob/e21397ce824c514e2796cbd21fef436b8f37e9d2/module/opcua/nodevalue/nodevalue.go#L222

darlandieterich, 2020-12-07:
I had found it, but forgot to close the issue. solved using `.Value()`

Thank you very much!

closed 2020-12-07 by darlandieterich