gopcua/opcua #415: Example Subscribe: sub.Monitor returns Status not ok

jon-ryan, 2021-02-08:
I'm new to opcua and following the "Subscribe" example. I need this functionality in my application later on. 
I followed everything step by step but in [line 95](https://github.com/gopcua/opcua/blob/4f8488dd4b82f886ee111bb33d1b9471f5b94a58/examples/subscribe/subscribe.go#L95) I always get a "status not ok". 

My opcua server (made in python) is up and running.

My OPCUA client settings are:
- Policy: None
- Mode: None
- Cert: <empty>
- KeyFile: <empty>
- NodeID: 85
- Event: true (also tried it with false but got same error)
- Interval: 1 second

I looked into the source cod of sub.Monitor but this only confused me even more.

Thank you for helping.

jon-ryan, 2021-02-16:
Update: I got it working passed the point described above. But now I don't receive any data on the notifyChan. I checked if my server is working correctly with an OPC-UA GUI and I can confirm that my server is running correctly.

bo-er, 2021-02-19:
I have the same problem.😭

bo-er, 2021-02-19:
And the solutions is add
```
go sub.Run(ctx)
```
before
```
for {
		select {
		case <-ctx.Done():
			return
		case res := <-notifyCh:
			fmt.Println("收到消息")
			if res.Error != nil {
				log.Print(res.Error)
				continue
			}

			switch x := res.Value.(type) {
			case *ua.DataChangeNotification:
				for _, item := range x.MonitoredItems {
					data := item.Value.Value.Value()
					log.Printf("MonitoredItem with client handle %v = %v", item.ClientHandle, data)
				}

			case *ua.EventNotificationList:
				for _, item := range x.Events {
					log.Printf("Event for client handle: %v\n", item.ClientHandle)
					for i, field := range item.EventFields {
						log.Printf("%v: %v of Type: %T", eventFieldNames[i], field.Value(), field.Value())
					}
					log.Println()
				}

			default:
				log.Printf("what's this publish result? %T", res.Value)
			}
		}
	}
```

magiconair, 2021-02-19:
Note that this will no longer be necessary when we merge v0.2

closed 2021-03-26 by magiconair