gopcua/opcua #422: How to reconnect after opcua is disconnected

beyondyinjl2, 2021-03-11:
The opcua collection program collects data after a normal connection. After the opcua service is suddenly shut down, the opcua collection program loses the connection. After the opcua service returns to normal, how to automatically reconnect.
The error is as follows:
write tcp 192.168.3.40:57545->192.168.3.48:53530: write: broken pipe
An error is reported when calling, Client.Connect(ctx):
opcua: already connected

magiconair, 2021-03-26:
The 0.2.0-rc1 version on master should automatically reconnect.

closed 2021-03-26 by magiconair

czwen, 2023-03-22:
should i renew the client(or what should i do) after reconnect successfully?
it still can't read value when state become connected from reconnecting.

2023/03/22 11:51:12 Read failed: %s write tcp 127.0.0.1:57007->127.0.0.1:4840: write: broken pipe
2023/03/22 11:51:12 write tcp 127.0.0.1:57007->127.0.0.1:4840: write: broken pipe