gopcua/opcua #425: Out of sequence chunk to the server when using multiple goroutines.

siscia, 2021-03-29:
I am using a server from the Python opcua package.

On the client, I got different goroutines looking for references of nodes:

```
nextNodes, err := node.ReferencedNodes(id.Organizes, ua.BrowseDirectionForward, ua.NodeClassObject, true)
```

Often, not always, the server raise this exception while the client hangs.

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/opcua/server/binary_server_asyncio.py", line 75, in _process_data
    ret = self.processor.process(hdr, buf)
  File "/usr/local/lib/python3.9/site-packages/opcua/server/uaprocessor.py", line 83, in process
    msg = self._connection.receive_from_header_and_body(header, body)
  File "/usr/local/lib/python3.9/site-packages/opcua/common/connection.py", line 310, in receive_from_header_and_body
    return self._receive(chunk)
  File "/usr/local/lib/python3.9/site-packages/opcua/common/connection.py", line 341, in _receive
    self._check_incoming_chunk(msg)
  File "/usr/local/lib/python3.9/site-packages/opcua/common/connection.py", line 290, in _check_incoming_chunk
    raise ua.UaError("Received chunk: {0} with wrong sequence expecting: {1}, received: {2}, spec says to close connection".format(chunk, self._peer_sequence_number, num))
opcua.ua.uaerrors._base.UaError: Received chunk: MessageChunk(Header(type:b'MSG', chunk_type:b'F', body_size:137, channel:6), SequenceHeader(SequenceNumber:490, RequestId:490 ), SymmetricAlgorithmHeader(TokenId:13 ), 125 bytes) with wrong sequence expecting: 329, received: 490, spec says to close connection
```

Is this error expected? Should I not use different goroutines?

With a single goroutines I was not able to reproduce the issue.
Is the connection safe to use in multiple goroutines?

magiconair, 2021-03-29:
We use multiple goroutines with a single connection all the time but I think there was a bug just fixed on master. Have a look and try that. 

—
Frank Schröder

siscia, 2021-03-30:
Can't replicate in master!

closed 2021-03-30 by siscia