gopcua/opcua #433: Provide feedback on connection status

magiconair, 2021-04-09:
In 0.2 we have auto-reconnection enabled by default which means that we should add some functions to determine the current connection status (e.g. `IsConnected`) and callbacks when we lose or re-establish the connection for monitoring purposes.

kung-foo, 2021-04-09:
For reference, I've used the NATS library quite extensively and they expose the connection in like this: https://pkg.go.dev/github.com/nats-io/nats.go#Conn.SetClosedHandler

(both via explict "SetXXX" calls, and "options" on the constructor)

magiconair, 2021-04-11:
That looks nice. They also have an option for a custom dialer which looks like a nicer approach than I've come up with in #432

closed 2025-04-26 by github-actions[bot]