gopcua/opcua #435: StatusBadResponseTooLarge during connection

mirkocomparetti-synesis, 2021-04-21:
Using the library to connect to an opcua server with telegraf, we get an error response with our server.
The server is a KUKA robot using the DeviceConnector. Python can connect fine after increasing `opcuaclient.max_messagesize = 2147483647` (2^31 - 1).
How can this be done with your library?
From our debug it looks like the problem comes in the response that is generated by the async call https://github.com/gopcua/opcua/blob/baabb9b14fd272417c63c27cfce625a16c519518/uasc/secure_channel.go#L763

I'm also linking the issue on the telegraf side https://github.com/influxdata/telegraf/issues/9163
Thanks!

codingismyAll, 2021-04-27:
How can we solve the same problem

codingismyAll, 2021-04-27:
change the  DefaultMaxMessageSize, the default value is 2mb, just set bigger

mirkocomparetti-synesis, 2021-04-27:
I did try to recompile with a larger default value, but it doesn't look like it helps. I most probably changed the value in the wrong place, I guess.
Though this should be configurable and set by a method call, IMHO

magiconair, 2021-09-16:
Can we get a tcpdump comparison between python and gopcua?

magiconair, 2021-09-23:
@mirkocomparetti-synesis I would like to try to address this for v0.2. Are you still working on this?

diericd, 2021-09-24:
Isn't this related to client->server message chuncking?

magiconair, 2021-11-09:
You can set the handshake parameters by providing a custom `uacp.Dialer` but I think we should provide a config option for this. 

```go
	opts := []opcua.Option{
		...
		opcua.Dialer(&uacp.Dialer{
			ClientACK: &uacp.Acknowledge{
				ReceiveBufSize: 8192,
				SendBufSize:    8192,
				MaxMessageSize: 2 * 1024 * 1024,
				MaxChunkCount:  512,
			},
		}),
	}

	c := opcua.NewClient(ep.EndpointURL, opts...)
	if err := c.Connect(ctx); err != nil {
		log.Fatal(err)
	}
	defer c.Close()
```

PR #496 (2021-11-11): Add handshake config options (#495)
This patch adds config options for the client handshake options

* MaxMessageSize
* MaxChunkCount
* ReceiveBufferSize
* SendBufferSize

Fixes #435
Fixes #495
closed 2021-11-11 by kung-foo