gopcua/opcua #437: How to start the opc server ?

mjrlgue, 2021-05-10:
I downloaded the package with the `go` command, when I want to run the server with  ` go run server.go`, I get the following error: `go run: cannot run non-main package`, I can't find a well-explained doc of how to start the server properly, can someone point me to the right doc or command pls ?

magiconair, 2021-05-11:
The server had not been implemented yet. Only the client package works

Duell10111, 2021-06-29:
Is there some timeline or plans to implement the server?

magiconair, 2021-06-30:
Not at this point. We had a couple of attempts but the biggest thing missing is the address space implementation.

magiconair, 2021-06-30:
PRs are welcome of course :)

closed 2021-06-30 by magiconair

heaxo, 2023-05-04:
@magiconair Hello, can the latest version achieve creating an Opcua server

magiconair, 2023-05-04:
No. This is still work in progress —Frank Schröder