gopcua/opcua #439: read.go example

rdockterjr, 2021-05-20:
I have a basic questions regarding the read.go example. I am trying to read a known tag from my plc. It looks like the database tag is the following: `Distance_Sensor_DB.Total_Count`

So I am running the following example:
`go run examples/read/read.go -endpoint opc.tcp://localhost:4840 -node "ns=3;s=""Distance_Sensor_DB"".""Total_Count"""`

However I get the following error:
`The node id refers to a node that does not exist in the server address space. StatusBadNodeIDUnknown (0x80340000)`

I can verify that the node exists. Am I doing something wrong as far as formatting the node argument goes?

magiconair, 2021-05-21:
Put single quotes around the node id. Otherwise the shell interprets them. 

—
Frank Schröder

rdockterjr, 2021-05-21:
Thanks alot. I figured it was something simple like that.

closed 2021-05-21 by rdockterjr