gopcua/opcua #448: msg.Value.String() returns empty string

mozesa, 2021-08-24:
Hello!

First, I would like to thank you implementing OPC UA protocol for go community.

I am at version: `v0.2.0-rc2` and I noticed that `msg.Value.String()` returns `""` using `monitor.go` example.

[image]

Actually, I would like to achieve to have the string representation of all the values.

Thanks for your help in advance.

magiconair, 2021-08-24:
Thank you!

I'm not sure I understand your question. Is the value a string and is it empty?

mozesa, 2021-08-24:
I subscribed to `xSensor` and `iSensor`.
[image]

`msg.Value.Value()` works fine and gives me: `true`, `11` but the types are `interface{}`. 

What is the purpose of the `msg.Value.String()` function?
I thought that it should return the `string` representation of any type instead of `interface{}`.

magiconair, 2021-08-24:
Yeah, it should probably do that but right now it returns the string value if the variant type is string. To achieve what you want you could do `fmt.Sprintf("%v", msg.Value.Value())`

https://github.com/gopcua/opcua/blob/main/ua/variant.go#L472-L491

closed 2021-08-24 by magiconair

mozesa, 2021-08-24:
Thanks! I actually do that 😄