gopcua/opcua #449: invalid value: opcua: cannot set variant to int

mozesa, 2021-08-24:
Hello!

I have run into the following error message: `invalid value: opcua: cannot set variant to int`.

I used the `write.go` example.
The `iSensor` node, as you can see, has an `INT` type.

(_Btw, setting the the value of `xSensor` works fine, with `ua.NewVariant(true)`._)

I am using Codesys soft PLC.

[image]

gopcua: `v0.2.0-rc2`
golang: `1.17`

Thanks for your help in advance.

magiconair, 2021-08-24:
yes, `int` and `uint` are not supported since they are platform dependent. Use any of the other types like `int32` or `uint16`

magiconair, 2021-08-24:
`int` and `uint` are either 32 or 64 bit depending on the CPU architecture.

closed 2021-08-24 by magiconair

mozesa, 2021-08-24:
Thanks for your help.
I leave it for the future.
[image]