gopcua/opcua #450: v0.2.0 official version

longxinH, 2021-08-28:
Hi，When will the official version of v0.2.0 be released?

magiconair, 2021-09-13:
I will try to make a push for making v0.2.0 production ready. When we tested it on some of our PLCs we crashed the Beckhoff OPC/UA server component. We also need connection status hooks to support metrics and alarms because the connection is now re-established in the background. I'll check if the reconnect logic is maybe a bit too aggressive and overloads some PLCs or at least hits some limits too quickly which can lead to loss of connection b/c the PLC won't allow us to connect for a while. 

Since the subscription logic has changed as well I would also like to make the number of outstanding publish requests configurable. Right now this is set to 1 but we might need more if we have only a single subscription loop.

So I can't give a date yet but we (Northvolt) have a need to move to v0.2.0 to overcome some limits. Stay tuned.

magiconair, 2021-09-16:
I'll close this. Please watch the 0.2 milestone.

closed 2021-09-16 by magiconair