gopcua/opcua #453: Data race in subscription mode

antlad, 2021-09-06:
Our CI pipeline detected data race in monitoring mode. Looks like it is happening after modifying list of nodes in monitoring.
gopcua/opcua commit `9bdbeb885f64`  from master.

```bash
WARNING: DATA RACE
Write at 0x00c00033a6b8 by goroutine 27:
  github.com/gopcua/opcua.(*Client).updatePublishTimeout()
      xxx/vendor/github.com/gopcua/opcua/client_sub.go:225 +0x64
  github.com/gopcua/opcua.(*Client).Subscribe()
      xxx/vendor/github.com/gopcua/opcua/client_sub.go:68 +0x9b9
  xxx/subscription.(*opcuaSubClient).connect()
      xxx/opcua/subscription/opcua_subscription.go:737 +0x4b6
DEVID:4c9694a1-a334-4dcf-bbbf-9737b34eb7ef =>   xxx/opcua/subscription.(*opcuaSubClient).Init()
      xxx/opcua/subscription/opcua_subscription.go:685 +0x5d
  xxx.(*device).Run()
      xxx:604 +0xb65

Previous read at 0x00c00033a6b8 by goroutine 41:
  github.com/gopcua/opcua.(*Client).sendPublishRequest()
      xxx/vendor/github.com/gopcua/opcua/client_sub.go:511 +0x2a4
  github.com/gopcua/opcua.(*Client).publish()
      xxx/vendor/github.com/gopcua/opcua/client_sub.go:372 +0x195
  github.com/gopcua/opcua.(*Client).monitorSubscriptions()
      xxx/vendor/github.com/gopcua/opcua/client_sub.go:354 +0x2d4
```

siscia, 2021-09-13:
Can you try again with the code on the PR?

closed 2021-09-16 by magiconair

antlad, 2021-09-16:
Looks like this gone. But found other one, soon will create issue

PR #455 (2021-09-16): safe concurrent access to publishTimeout #453
Somehow ugly, but it should take care of #453 

Fixes #453
commit 6033c33b (2021-09-16): safe concurrent access to publishTimeout #453

Fixes #453