gopcua/opcua #458: Clear Subscription item when Unmonitor

tomhollingworth, 2021-09-15:
I just noticed I there are still references a `monitoredItem` even after `Unmonitor` is called on the `monitoredItemIDs`. Should items be cleaned up in `Unmonitor`?

I'm happy to submit a PR.

For example:

subscription.go
```diff golang

func (s *Subscription) Unmonitor(monitoredItemIDs ...uint32) (*ua.DeleteMonitoredItemsResponse, error) {
	req := &ua.DeleteMonitoredItemsRequest{
		MonitoredItemIDs: monitoredItemIDs,
		SubscriptionID:   s.SubscriptionID,
	}
	var res *ua.DeleteMonitoredItemsResponse
	err := s.c.Send(req, func(v interface{}) error {
		return safeAssign(v, &res)
	})
+
+	if err == nil {
+              // cleanup items if no error
+		index := 0
+		for _, item := range s.items {
+			if !contains(monitoredItemIDs, item.createResult.MonitoredItemID) {
+				s.items[index] = item
+				index++
+			}
+		}
+		for i := index; i < len(s.items); i++ {
+			s.items[i] = nil
+		}
+		s.items = s.items[:index]
+	}
+
	return res, err
}

+// contains returns true if the array contains the search term
+func contains(array []uint32, search uint32) bool {
+	for _, i := range array {
+		if i == search {
+			return true
+		}
+	}
+	return false
+}
```

magiconair, 2021-09-16:
This sounds off. I'll have a look

magiconair, 2021-09-25:
@tomhollingworth could you please check #471? It is basically your change but without modifying the slice in-place.

commit d11cd844 (2021-11-08): remove monitored items on Unmonitor

Fixes #458
closed 2021-11-08 by magiconair