gopcua/opcua #460: Data race in security channel timeout

antlad, 2021-09-16:
One more data race when when adding option opcua.Lifetime(someDuration)  when it is time to renew. Happening with Prosys simulator server, but not with Kepware.

```
==================
WARNING: DATA RACE
Write at 0x00c0004dc180 by goroutine 47:
  github.com/gopcua/opcua/uasc.(*SecureChannel).reset()
      xxx/vendor/github.com/gopcua/opcua/uasc/secure_channel.go:174 +0x25c
  github.com/gopcua/opcua/uasc.(*SecureChannel).Close.func1()
      xxx/vendor/github.com/gopcua/opcua/uasc/secure_channel.go:839 +0x5e
  github.com/gopcua/opcua/uasc.(*SecureChannel).Close()
      xxx/vendor/github.com/gopcua/opcua/uasc/secure_channel.go:847 +0x2d4
  github.com/gopcua/opcua.(*Client).monitor()
      xxx/vendor/github.com/gopcua/opcua/client.go:323 +0x1364

Previous read at 0x00c0004dc180 by goroutine 48:
  github.com/gopcua/opcua/uasc.(*SecureChannel).popHandler()
      xxx/vendor/github.com/gopcua/opcua/uasc/secure_channel.go:732 +0xcc
  github.com/gopcua/opcua/uasc.(*SecureChannel).sendRequestWithTimeout()
      xxx/vendor/github.com/gopcua/opcua/uasc/secure_channel.go:712 +0x4e5
  github.com/gopcua/opcua/uasc.(*SecureChannel).SendRequestWithTimeout()
      xxx/vendor/github.com/gopcua/opcua/uasc/secure_channel.go:760 +0x135
  github.com/gopcua/opcua.(*Client).sendWithTimeout()
      xxx/vendor/github.com/gopcua/opcua/client.go:764 +0xf5
  github.com/gopcua/opcua.(*Client).sendPublishRequest()
      xxx/vendor/github.com/gopcua/opcua/client_sub.go:512 +0x316
  github.com/gopcua/opcua.(*Client).publish()
      xxx/vendor/github.com/gopcua/opcua/client_sub.go:373 +0x18b
  github.com/gopcua/opcua.(*Client).monitorSubscriptions()
      xxx/vendor/github.com/gopcua/opcua/client_sub.go:355 +0x2c9

Goroutine 47 (running) created at:
  github.com/gopcua/opcua.(*Client).Connect.func1()
      xxx/vendor/github.com/gopcua/opcua/client.go:213 +0x7a
  sync.(*Once).doSlow()
      /Users/xxx/sdk/go1.14.15/src/sync/once.go:66 +0x103
  sync.(*Once).Do()
      /Users/xxx/sdk/go1.14.15/src/sync/once.go:57 +0x68
  github.com/gopcua/opcua.(*Client).Connect()
      xxx/vendor/github.com/gopcua/opcua/client.go:212 +0x27c
  
Goroutine 48 (running) created at:
  github.com/gopcua/opcua.(*Client).Connect.func1()
      xxx/vendor/github.com/gopcua/opcua/client.go:214 +0xb0
  sync.(*Once).doSlow()
      /Users/xxx/sdk/go1.14.15/src/sync/once.go:66 +0x103
  sync.(*Once).Do()
      /Users/xxx/sdk/go1.14.15/src/sync/once.go:57 +0x68
  github.com/gopcua/opcua.(*Client).Connect()
      xxx/vendor/github.com/gopcua/opcua/client.go:212 +0x27c
==================

```

magiconair, 2021-09-17:
Issue isn't fixed yet

magiconair, 2021-09-24:
races are gone for now.

magiconair, 2021-09-24:
Oh, this one might be different.

antlad, 2021-09-24:
@magiconair Thanks for last updates. Looks like this one gone (at least i don't see it in same conditions as before). I was trying with Prosys demo server and manipulated lifetime of security channel.

closed 2021-09-24 by magiconair