gopcua/opcua #463: Error writing byte(uint8) arrays. Wrong array length.

SkyFox277, 2021-09-20:
Hello,
I read some Data from server and get a variant with this properties:
  ```
  mask: 131
  arrayLength: 54
  arrayDimensionsLength: 0
  arrayDimensions: nil
  value: interface{} ([]uint8) [......]
  ```
If I try to write this data back
  ```
  var buf []byte = make([]byte, 54)
  buf = (*(*[54]byte)(unsafe.Pointer(&wrVal)))[:]
  v, err := ua.NewVariant(buf)
```
At this point the new variant has this properties:
```
  mask: 15
  arrayLength: 0
  arrayDimensionsLength: 0
  arrayDimensions: nil
  value: interface{} ([]uint8) [......]
```
I get: "The value supplied for the attribute is not of the same type as the attribute's value. StatusBadTypeMismatch (0x80740000)"

This issue is caused by variang.go (row: 412-415)

kung-foo, 2021-09-20:
I am not able to reproduce.

If I tweak the `examples/write` demo:

```patch
diff --git a/examples/write/write.go b/examples/write/write.go
index 8eb35e5..535b0e2 100644
--- a/examples/write/write.go
+++ b/examples/write/write.go
@@ -18,7 +18,6 @@ func main() {
        var (
                endpoint = flag.String("endpoint", "opc.tcp://localhost:4840", "OPC UA Endpoint URL")
                nodeID   = flag.String("node", "", "NodeID to read")
-               value    = flag.String("value", "", "value")
        )
        flag.BoolVar(&debug.Enable, "debug", false, "enable debug logging")
        flag.Parse()
@@ -37,7 +36,7 @@ func main() {
                log.Fatalf("invalid node id: %v", err)
        }
 
-       v, err := ua.NewVariant(*value)
+       v, err := ua.NewVariant([]byte{6, 7, 8, 9, 10})
        if err != nil {
                log.Fatalf("invalid value: %v", err)
        }
```

It works as expected:

[image]

Though I am a bit unsure as to the `unsafe.Pointer` bits in your report...

SkyFox277, 2021-09-21:
This is my minimal example:
```
package main

import (
	"bytes"
	"context"
	"encoding/binary"
	"flag"
	"log"

	"github.com/gopcua/opcua"
	"github.com/gopcua/opcua/debug"
	"github.com/gopcua/opcua/ua"
)

type myStruct struct {
	WdEcn     uint16
	RxLen     uint16
	ErrorCode uint16
	Connect   bool
	Activity  bool
	Dummy     bool
	Error     bool
	RxTime    [32]byte
	RxTimeArr [6]uint16
}

func main() {
	var (
		endpoint = flag.String("endpoint", "opc.tcp://192.168.178.210:4840", "OPC UA Endpoint URL")
		nodeID   = flag.String("node", "ns=2;s=0:TCE_1.write", "NodeID to write")
	)
	flag.BoolVar(&debug.Enable, "debug", false, "enable debug logging")
	flag.Parse()
	log.SetFlags(0)

	ctx := context.Background()

	c := opcua.NewClient(*endpoint)
	if err := c.Connect(ctx); err != nil {
		log.Fatal(err)
	}
	defer c.Close()

	id, err := ua.ParseNodeID(*nodeID)
	if err != nil {
		log.Fatalf("invalid node id: %v", err)
	}

	// read
	rreq := &ua.ReadRequest{
		MaxAge: 2000,
		NodesToRead: []*ua.ReadValueID{
			&ua.ReadValueID{NodeID: id},
		},
		TimestampsToReturn: ua.TimestampsToReturnBoth,
	}

	rdVal := myStruct{}
	resp, err := c.Read(rreq)
	if err != nil {
		log.Fatalf("Read failed: %s", err)
	}
	if resp.Results[0].Status != ua.StatusOK {
		log.Fatalf("Status not OK: %v", resp.Results[0].Status)

	} else {
		// **** Variant ****
		// mask: 131
		// arrayLength: 54
		// arrayDimensionsLength: 0
		// arrayDimensions: nil
		// value: interface{} ([]uint8) [......]

		r := bytes.NewReader(resp.Results[0].Value.Value().([]byte))

		if err := binary.Read(r, binary.LittleEndian, &rdVal); err != nil {
			log.Printf("Binary read failed. err=%v", err)
		}
	}

	wrVal := myStruct{
		WdEcn: 90,
	}

	// var buf []byte = make([]byte, 54)
	// buf = (*(*[54]byte)(unsafe.Pointer(&wrVal)))[:]

	// v, err := ua.NewVariant(buf)
	// if err != nil {
	// 	log.Fatalf("invalid value: %v", err)
	// }

	var bin_buf bytes.Buffer
	binary.Write(&bin_buf, binary.LittleEndian, wrVal)

	v, err := ua.NewVariant(bin_buf.Bytes())
	if err != nil {
		log.Fatalf("invalid value: %v", err)
	}

	// **** Variant ****
	// mask: 15
	// arrayLength: 0
	// arrayDimensionsLength: 0
	// arrayDimensions: nil
	// value: interface{} ([]uint8) [......]

	wreq := &ua.WriteRequest{
		NodesToWrite: []*ua.WriteValue{
			&ua.WriteValue{
				NodeID:      id,
				AttributeID: ua.AttributeIDValue,
				Value: &ua.DataValue{
					EncodingMask: ua.DataValueValue,
					Value:        v,
				},
			},
		},
	}

	wrresp, err := c.Write(wreq)
	if err != nil {
		log.Fatalf("Write failed: %s", err)
	}
	log.Printf("write resp=%v", wrresp.Results[0])
}
```
It may be that it depends on the server used. But the variant property arrayLength is definitely wrong. And in my case the server does not allow to write this value.
```
	v1, err := ua.NewVariant([]uint16{6, 7, 8, 9, 10})
	log.Printf("variant: %#v", v1)
	v2, err := ua.NewVariant([]uint8{6, 7, 8, 9, 10})
	log.Printf("variant: %#v", v2)
	v3, err := ua.NewVariant([]byte{6, 7, 8, 9, 10})
	log.Printf("variant: %#v", v3)
	// get:
	// variant: &ua.Variant{mask:0x85, arrayLength:5, arrayDimensionsLength:0, arrayDimensions:[]int32(nil), value:[]uint16{0x6, 0x7, 0x8, 0x9, 0xa}}
	// variant: &ua.Variant{mask:0xf, arrayLength:0, arrayDimensionsLength:0, arrayDimensions:[]int32(nil), value:[]uint8{0x6, 0x7, 0x8, 0x9, 0xa}}
	// variant: &ua.Variant{mask:0xf, arrayLength:0, arrayDimensionsLength:0, arrayDimensions:[]int32(nil), value:[]uint8{0x6, 0x7, 0x8, 0x9, 0xa}}
```

kung-foo, 2021-09-21:
OK. I think I understand the issue now. When testing against ProSys, it seems to accept a write request with `TypeIDByteString` even though the server type is `TypeIDByte`.

The underlying issue seems to be that since `TypeIDByteString` and `TypeIDByte` are both represented by `[]byte` in go, we default to a `TypeIDByteString`.

We are correctly encoding a `ByteString` as `arrayLength:0, arrayDimensionsLength:0, arrayDimensions:[]int32(nil)`. I verified that this is also what ProSys is _returning_ for a `ByteString`. When I read a `ByteArray` from ProSys I get `arrayLength:5, arrayDimensionsLength:0, arrayDimensions:[]int32(nil)`.

So for your server, we need a way to force a variant to be `TypeIDByte` rather than `TypeIDByteString`.

Not sure how to proceed for right now. Possibly introducing either a type alias (i.e. `type ByteString []byte`) and changing the default behavior for `[]byte` to be `TypeIDByte`. Or possibly an explicit constructor like `NewByteStringVariant([]byte)`.

Note, if you just comment out https://github.com/gopcua/opcua/blob/v0.2.0-rc2/ua/variant.go#L413-L415 then you effectively change the behavior to `TypeIDByte`. However, then `TypeIDByteString` is not possible, and ProSys does _not_ handle writing a `TypeIDByte` to a `TypeIDByteString`.

/cc @magiconair

kung-foo, 2021-09-21:
For reference: https://reference.opcfoundation.org/v104/Core/docs/Part6/5.1.4/

magiconair, 2021-09-21:
I knew this was going to bite us at some point. I'll look into this.

magiconair, 2021-09-23:
@SkyFox277 what do you mean by that:

> In my opinion the best solution would be to check whether the []byte contains the length and whether this length is correct and only in this case to set the type to TypeIDByteString.

magiconair, 2021-09-23:
I never understood why they have `String`, `Byte` and `ByteString`. What use-case does this support?

https://reference.opcfoundation.org/v104/Core/docs/Part6/5.1.4/ 

>A ByteString is structurally the same as a one-dimensional array of Byte. It is represented as a distinct built-in data type because it allows encoders to optimize the transmission of the value. However, some DevelopmentPlatforms will not be able to preserve the distinction between a ByteString and a one-dimensional array of Byte.
> 
> If a decoder for DevelopmentPlatform cannot preserve the distinction it shall convert all one-dimensional arrays of Byte to ByteStrings.

magiconair, 2021-09-23:
I remember wrestling with this since a `type ByteString []byte` seemed unnecessary. If we would add this now *and* change the default we would be breaking existing apps in a non-obvious way. This is different from an API change that the compiler can detect.

What about adding `type ByteArray []byte` and `type ByteString []byte` and leave the default at `ByteString` for `[]byte` since this is what the spec seems to suggest?

magiconair, 2021-09-23:
I've added a `type ByteArray []byte` to the `Variant` and tested single and multi-dimensional array support. See if this works.

PR #467 (2021-09-24): Add support for ByteArray
This patch adds support for the ByteArray type in a Variant.
This allows the caller to send a []byte as an array of Byte instead of a
length encoded ByteString which is the default.

Fixes #463

/cc @SkyFox277
commit 7302344e (2021-09-24): Add support for ByteArray

This patch adds support for the ByteArray type in a Variant.
This allows the caller to send a []byte as an array of Byte instead of a
length encoded ByteString which is the default.

Fixes #463
closed 2021-09-24 by kung-foo