gopcua/opcua #464: Allow Timeout when Listing Endpoints/Dial

Mendes11, 2021-09-20:
Hi and thanks for the amazing library!

When my application tries to List all Endpoints of a specific server URL that server doesn't exists, it freezes for some time until I receive a 'connection timed out' error.

Is it possible to configure a Timeout so I won't have to wait for too long?

I Tried to create a new Client with `SessionTimeout` and `RequestTimeout` options and then calling Dial, but it ignored those values. 

How it Is:

```go
opcua.GetEndpoints(server.URL)
```

How I propose it could be:

```go
opts := []opcua.DialOption{
    opcua.Timeout(time.Seconds * 2),
}
opcua.GetEndpoints(server.URL, opts...)
```

siscia, 2021-09-22:
Maybe it would be better to accept a context?

magiconair, 2021-09-22:
Yes let's add that

siscia, 2021-09-22:
But maybe I am missing something: https://github.com/gopcua/opcua/blob/main/client.go#L29

siscia, 2021-09-22:
Aw maybe @Mendes11 refers to the client method!

https://github.com/gopcua/opcua/blob/8fbaa57910cd4ab26ab0b80c7e6d0c029b6d823e/client.go#L788

That indeed does not accept a context.

I believe it would be quite useful to add the context down to `client.Send()`.
That all the other functions are based upon, including `GetEndpoints`.

I believe it would be rather simple to do.

siscia, 2021-09-22:
But we will break the public API...

Mendes11, 2021-09-22:
> But we will break the public API...

You mean regarding `GetEndpoints` function?

Since that would break the API, what about when I do what `GetEndpoints` function do?

I Tried to:
```go
opts := []opcua.Option{opcua.RequestTimeout(time.Second), opcua.SessionTimeout(time.Second)}
c := opcua.NewClient(server.URL,  opts...)
if err := c.Dial(context.Background()); err != nil {
	log.Panic(err)
}
res, err := c.GetEndpoints()
if err != nil {
    log.Panic(err)
}

log.Println(res.Endpoints)
```

I Would expect that this Timeout options would make my code wait for only 1 second but it doesn't do that...

magiconair, 2021-09-22:
I'm not worried about breaking the API. We're at v0.1 moving to v0.2. There is no guarantee for stability at this point. More important to get this right. I suggest to just use `GetEndpoints(ctx)`

magiconair, 2021-09-22:
I think I've already addressed this for v0.2 ...

[image]

magiconair, 2021-09-22:
I need to look at this a bit closer. `c.GetEndpoints()` should honour the `RequestTimeout`.

Mendes11, 2021-09-22:
> I need to look at this a bit closer. `c.GetEndpoints()` should honour the `RequestTimeout`.

It doesn't because of `uacp.Dial`

Looking through the source code, the function itself doesn't handle timeouts:

uacp/conn.go: 
```go
func Dial(ctx context.Context, endpoint string) (*Conn, error) {
	debug.Printf("Connect to %s", endpoint)
        ....
```

Since all methods, including `Connect`, first call  `Dial` before actually connecting, the timeout configurations will never work for first connections, only when `SecureChannel` is stabilished that they actually work, due to the configurations only being used there.

Mendes11, 2021-09-22:
From 
https://github.com/gopcua/opcua/blob/d297b502ea2e4a93f249a730bb3d4727f138ce3e/client.go#L476

we can see that `c.cfg` is pased only after Dial was called (To `SecureChannel`)

magiconair, 2021-09-23:
@Mendes11 Could you please have a look at #468 ? I've created the `uacp.Dialer` a while ago but did not make it configurable in the client yet. I've refactored the config options to include the dialer and also add a `DialTimeout` option which defaults to the `RequestTimeout`. I've also added the options to the `GetEndpoints` call to make sure that the options are applied. This should fix your problem and provide more control for others.

Mendes11, 2021-09-23:
> @Mendes11 Could you please have a look at #468 ? I've created the `uacp.Dialer` a while ago but did not make it configurable in the client yet. I've refactored the config options to include the [...]

That seems to fix my problem indeed!

PR #468 (2021-09-24): Make dial timeout configurable
This patch makes the `uacp.Dialer` configurable and allows to configure the dial timeout for establishing the uucp connection.

Fixes #464
closed 2021-09-24 by magiconair