gopcua/opcua #472: Too many sessions

siscia, 2021-09-27:
While keeping the number of connections to the OPCUA server constants, I keep seeing errors like:

```
10:45:23.460078 The server has reached its maximum number of sessions. StatusBadTooManySessions (0x80560000)
10:45:23.877245 The server has reached its maximum number of sessions. StatusBadTooManySessions (0x80560000)
```

This error is well managed in the code, but maybe it should be fixed at the library level.

It seems like when a OPCUA connection is disconnected, the session it is not closed immediately.
Now, I am not sure if this is wanted, or not.

milgner, 2021-09-29:
What server are you using? I just encountered the selfsame error with a different library where sessions were kept on the server even after the client disconnected.

siscia, 2021-09-29:
I am mostly agains Siemens equipment, but I am not sure of the server details.

milgner, 2021-09-29:
The problem usually only occurs if the client doesn't send a CloseSession request. Check your code: it should have `defer client.Close()` right after establishing the connection or you will need to manually close it wherever appropriate.

siscia, 2021-09-29:
Thanks a lot!

We are already closing the session with a call to `client.Close()` which is indeed the reason why I opened the ticket.

milgner, 2021-09-29:
In that case I am out of ideas. I encountered the same problem with another client/server combination but with this library everything works expected and the session is gone after closing the client.

fenogentov, 2022-02-09:
close() does not close the channel and not reset the session

closed 2025-04-26 by github-actions[bot]