gopcua/opcua #494: Could not resolve address ...

mozesa, 2021-11-08:
Dear **@magiconair**,

I would like to ask for your help.
I have the following error message: `could not resolve address tben-lx-plc:4840`

I have enclosed an image which depicts the endpoints, which contains 6 endpoints.
The problem that endpoints[0] always gives the error: `could not resolve ...`.
But I could use endpoints[3] to connect successfully.

Is there any method by which I can filter out endpoints which EndpointURL like "tben-lx-plc".

[image]

Thanks for your help in advance.

kung-foo, 2021-11-08:
It's pretty common for an opc-ua server to just return the hostname of itself. That may or may not be resolvable as is by your client. Sometime you need to append a domain. Sometimes it returns a private name.

mozesa, 2021-11-08:
I see! But is there a standard way to be able to connect to such an OPC UA server?

kung-foo, 2021-11-08:
For our applications at my company, we rarely, if ever, use the `GetEndpoints` API. Or if we do, we _don't_ trust/use the hostname since it is not configured properly most of the time.

kung-foo, 2021-11-08:
In your case, it depends whether `tben-lx-plc` is the same as `192.168.0.21`

mozesa, 2021-11-08:
Yes, it is the same.
What should I use instead?

kung-foo, 2021-11-08:
Yes, this is what I use 99% of the time.

mozesa, 2021-11-08:
Thanks for the advice - it works!

closed 2021-11-08 by mozesa