gopcua/opcua #498: Keep Alive

NieYanzhai, 2021-11-19:
Is there opcua keepalive event handler?

or excample?

magiconair, 2021-11-19:
I usually subscribe to i=2258 which is the current time.

NieYanzhai, 2021-11-22:
I think client will close the session connection to server when server is offline or network failure, and keepalive will nofity this situation. But i don't know how to do this keepalive event using golang. Is there any excample?

magiconair, 2021-11-22:
Check the `examples/subscribe/subscribe.go` example for setting up a subscription. You don't need the `Monitor` for this. Just a simple subscription for that node `i=2258` should be enough.

NieYanzhai, 2021-11-23:
Thanks, Solved reconnect problem. Cool

2021/11/23 17:12:59 55
2021/11/23 17:13:00 87
2021/11/23 17:13:01 51
2021/11/23 17:13:02 disconnected
2021/11/23 17:13:03 disconnected
2021/11/23 17:13:10 35
2021/11/23 17:13:11 67
2021/11/23 17:13:12 99

closed 2021-11-23 by NieYanzhai