gopcua/opcua #503: The GOPCUA client crashed when the OPCUA server was restarted?

DanielZYH, 2021-11-19:
Hi, experts.
 I am running GOPCUA client (gopcua@v0.2.1/example/monitor/monitor.go) with OPCUA server(Siemens simatic WinccOA3.17 OPCUA server). When I restart(shutdown and re-open) OPCUA server, the GOPCUA client crashed, and printed out the following logs:
INFO: 2021/11/10 20:09:14 monitor_zyh.go:272: [channel 0] counter=166000
panic: runtime error: invalid memory address or nil pointer dereference 
[signal 0xc0000005 code=0x0 addr=0x18 pc=0x5aa5f6]
 
goroutine 13 [running]:
github.com/gopcua/opcua.(*Client).ActivateSession(0xc0001c81a0, 0x0, 0x0, 0x0)
        D:/GoWorks/pkg/mod/github.com/gopcua/opcua@v0.2.1/client.go:649 +0x56
github.com/gopcua/opcua.(*Client).monitor(0xc0001c81a0, 0x711200, 0xc00003d800)
        D:/GoWorks/pkg/mod/github.com/gopcua/opcua@v0.2.1/client.go:345 +0xc66
created by github.com/gopcua/opcua.(*Client).Connect.func1
        D:/GoWorks/pkg/mod/github.com/gopcua/opcua@v0.2.1/client.go:193 +0x6a
 
 
My question:
1. What can I do to solve this problem?
 
 
 
2. There is no exception(OPCUA server shutdown) raised in gopcua@v0.2.1, although I already have set "SetErrorHandler" to client monitor handle.
But, I can  catch this exception in gopcua@v0.1.13 by "SetErrorHandler", and reconnect OPCUA server, monitor node.
What should I do to catch this situation in the client side with gopcua@v0.2.1?

magiconair, 2021-12-06:
This sounds like a bug in the connection and error handling. Can you check if this still happens with v0.2.3?

DanielZYH, 2021-12-08:
hi, magiconair
    I tried this issue again, but it was still raised a exception  for  opcua client (gopcua@v0.2.3/example/monitor/monitor.go) with OPCUA server(Siemens simatic WinccOA3.17 OPCUA server).
    The panic error is as followed:
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x18 pc=0x5aa536]

goroutine 27 [running]:
github.com/gopcua/opcua.(*Client).ActivateSession(0xc0001821a0, 0x0, 0x0, 0x0)
        D:/GoWorks/pkg/mod/github.com/gopcua/opcua@v0.2.3/client.go:658 +0x56
github.com/gopcua/opcua.(*Client).monitor(0xc0001821a0, 0x711000, 0xc00015e480)
        D:/GoWorks/pkg/mod/github.com/gopcua/opcua@v0.2.3/client.go:354 +0xc79
created by github.com/gopcua/opcua.(*Client).Connect.func1
        D:/GoWorks/pkg/mod/github.com/gopcua/opcua@v0.2.3/client.go:202 +0x6a
   

PS: I can share my test code for investigating this issue if you would like.

magiconair, 2021-12-08:
Yes please. Find me on Keybase if you want to share it in private.

magiconair, 2021-12-08:
If I test this with the `./uatest/rw_server.py` and `go run ./examples/monitor/monitor.go -node i=2258` I can CTRL-C the server and start it again without problems.

DanielZYH, 2021-12-10:
Sorry. for the late response.
I have tried this case with Prosys OPC UA Simulation Server. After I close the Prosys opcua server and wait one second(about 15 seconds), the problem is being reproduced. So, I guess this is a bug for mature opcua server application.
The logs are as follows:
2021/12/10 20:07:20 begn to connect OPCUA server
INFO: 2021/12/10 20:07:20 monitor_zyh.go:108: begin to connect OPCUA server
2021/12/10 20:07:20 conected to opcua server
INFO: 2021/12/10 20:07:20 monitor_zyh.go:131: connected to opcua server opc.tcp:
//windows10.microdone.cn:53530/OPCUA/SimulationServer
2021/12/10 20:07:20 [startChanSub] subIdx=0 node count=3
INFO: 2021/12/10 20:07:20 monitor_zyh.go:186: [startChanSub] subIdx=0 node count
=3
2021/12/10 20:07:21 [startCanSub 0] success sub=3
INFO: 2021/12/10 20:07:21 monitor_zyh.go:253: [startChanSub 0] success sub=3
2021/12/10 20:07:28 [chnnel 0] counter=10
INFO: 2021/12/10 20:07:28 monitor_zyh.go:272: [channel 0] counter=10
2021/12/10 20:07:37 [chnnel 0] counter=20
INFO: 2021/12/10 20:07:37 monitor_zyh.go:272: [channel 0] counter=20
2021/12/10 20:07:46 [chnnel 0] counter=30
INFO: 2021/12/10 20:07:46 monitor_zyh.go:272: [channel 0] counter=30
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x18 pc=0x5aa536]

goroutine 50 [running]:
github.com/gopcua/opcua.(*Client).ActivateSession(0xc00017e270, 0x0, 0x0, 0x0)
        D:/GoWorks/pkg/mod/github.com/gopcua/opcua@v0.2.3/client.go:658 +0x56
github.com/gopcua/opcua.(*Client).monitor(0xc00017e270, 0x710c40, 0xc000246000)
        D:/GoWorks/pkg/mod/github.com/gopcua/opcua@v0.2.3/client.go:354 +0xc79
created by github.com/gopcua/opcua.(*Client).Connect.func1
        D:/GoWorks/pkg/mod/github.com/gopcua/opcua@v0.2.3/client.go:202 +0x6a
exit status 2

My test env are:
1. win10
2. monitor.go(modified by me, added 3 nodes for channel subscrible.)
3. Prosys OPC UA Simulation Server is a free software, you can get it from the internet. Adding 3 nodes(integer) into the simulation part.
 
My test step:
1. Start monitor.go 
2. Start Prosys opcua server
3. To close the Prosys app after the subscribed data is received.
4. Wait a while for 15-30 seconds.
5. Start Prosys app
6. Check out the monitor console.
[monitor_zyh.zip](https://github.com/gopcua/opcua/files/7692708/monitor_zyh.zip)
The attached file is my test code.

Thank you for your help.

DanielZYH, 2021-12-13:
Anything could I provide?
I would like to provide what helps solve this issue.

antlad, 2021-12-27:
Exploring same issues on latest tag and master. And this is happening when we are trying self implemented reconnect logic(why self implemented - options opcua.AutoReconnect doesn't work).  At some point it crashing because of `sechan` is nil. Some new race conditions. I tried to fix on my fork: protected `sechan` with mutex (with connection as well), but whole logic stopped working - just not connecting.

magiconair, 2022-01-03:
@antlad I'm back from vacation and can have a look now. Any reliable mechanism to reproduce this helps.

magiconair, 2022-01-03:
OK, I can reproduce it.

[image]

magiconair, 2022-01-03:
OK, found it. The problem is not the secure channel but the session. On first disconnect the session gets detached and the code tries to activate it again but fails. Then on subsequent reconnects it tries to activate a nil session and that's where it crashes. 

Need to figure out how to fix this properly.

commit c49ed729 (2022-01-03): client: do not clear session during reconnect

This patch checks if the session to be restored exists before trying to
restore it. It also prevents the internal session variable from being
cleared during a reconnect since the client would then try to restore
the session only once.

Fixes #503
PR #534 (2022-01-03): client: do not clear session during reconnect
This patch checks if the session to be restored exists before trying to
restore it. It also prevents the internal session variable from being
cleared during a reconnect since the client would then try to restore
the session only once.

Fixes #503
closed 2022-01-03 by magiconair