gopcua/opcua #510: 0.2.2

fenogentov, 2021-11-22:
` ~/go/pkg/mod/github.com/gopcua/opcua@v0.2.1 > go test ./...
ok  	github.com/gopcua/opcua	0.036s
?   	github.com/gopcua/opcua/cmd/id	[no test files]
?   	github.com/gopcua/opcua/cmd/service	[no test files]
?   	github.com/gopcua/opcua/cmd/service/goname	[no test files]
?   	github.com/gopcua/opcua/cmd/status	[no test files]
?   	github.com/gopcua/opcua/debug	[no test files]
ok  	github.com/gopcua/opcua/errors	0.009s
?   	github.com/gopcua/opcua/examples/accesslevel	[no test files]
?   	github.com/gopcua/opcua/examples/browse	[no test files]
?   	github.com/gopcua/opcua/examples/crypto	[no test files]
?   	github.com/gopcua/opcua/examples/datetime	[no test files]
?   	github.com/gopcua/opcua/examples/endpoints	[no test files]
?   	github.com/gopcua/opcua/examples/history-read	[no test files]
?   	github.com/gopcua/opcua/examples/method	[no test files]
?   	github.com/gopcua/opcua/examples/monitor	[no test files]
?   	github.com/gopcua/opcua/examples/read	[no test files]
?   	github.com/gopcua/opcua/examples/regread	[no test files]
?   	github.com/gopcua/opcua/examples/server	[no test files]
?   	github.com/gopcua/opcua/examples/subscribe	[no test files]
?   	github.com/gopcua/opcua/examples/translate	[no test files]
?   	github.com/gopcua/opcua/examples/trigger	[no test files]
?   	github.com/gopcua/opcua/examples/udt	[no test files]
?   	github.com/gopcua/opcua/examples/write	[no test files]
?   	github.com/gopcua/opcua/id	[no test files]
?   	github.com/gopcua/opcua/monitor	[no test files]
ok  	github.com/gopcua/opcua/ua	0.015s
ok  	github.com/gopcua/opcua/uacp	2.082s
ok  	github.com/gopcua/opcua/uapolicy	1.012s
ok  	github.com/gopcua/opcua/uasc	0.018s
`

fenogentov, 2021-11-22:
`~/go/pkg/mod/github.com/gopcua/opcua@v0.2.2 > go test ./...
ok  	github.com/gopcua/opcua	(cached)
?   	github.com/gopcua/opcua/cmd/id	[no test files]
?   	github.com/gopcua/opcua/cmd/service	[no test files]
?   	github.com/gopcua/opcua/cmd/service/goname	[no test files]
?   	github.com/gopcua/opcua/cmd/status	[no test files]
?   	github.com/gopcua/opcua/debug	[no test files]
ok  	github.com/gopcua/opcua/errors	(cached)
?   	github.com/gopcua/opcua/examples/accesslevel	[no test files]
?   	github.com/gopcua/opcua/examples/browse	[no test files]
?   	github.com/gopcua/opcua/examples/crypto	[no test files]
?   	github.com/gopcua/opcua/examples/datetime	[no test files]
?   	github.com/gopcua/opcua/examples/endpoints	[no test files]
?   	github.com/gopcua/opcua/examples/history-read	[no test files]
?   	github.com/gopcua/opcua/examples/method	[no test files]
?   	github.com/gopcua/opcua/examples/monitor	[no test files]
?   	github.com/gopcua/opcua/examples/read	[no test files]
?   	github.com/gopcua/opcua/examples/regread	[no test files]
?   	github.com/gopcua/opcua/examples/server	[no test files]
?   	github.com/gopcua/opcua/examples/subscribe	[no test files]
?   	github.com/gopcua/opcua/examples/translate	[no test files]
?   	github.com/gopcua/opcua/examples/trigger	[no test files]
?   	github.com/gopcua/opcua/examples/udt	[no test files]
?   	github.com/gopcua/opcua/examples/write	[no test files]
?   	github.com/gopcua/opcua/id	[no test files]
?   	github.com/gopcua/opcua/monitor	[no test files]
runtime: goroutine stack exceeds 1000000000-byte limit
runtime: sp=0xc0203603b8 stack=[0xc020360000, 0xc040360000]
fatal error: stack overflow

runtime stack:
runtime.throw({0x5a2e84, 0x7773c0})
	/usr/lib/go-1.17/src/runtime/panic.go:1198 +0x71
runtime.newstack()
	/usr/lib/go-1.17/src/runtime/stack.go:1088 +0x5ac
runtime.morestack()
	/usr/lib/go-1.17/src/runtime/asm_amd64.s:461 +0x8b

goroutine 535 [running]:
fmt.(*pp).handleMethods(0xc0052abe10, 0x76)
	/usr/lib/go-1.17/src/fmt/print.go:570 +0x708 fp=0xc0203603c8 sp=0xc0203603c0 pc=0x4ac808
fmt.(*pp).printArg(0xc0052abe10, {0x59ad00, 0xc0002ff240}, 0x76)
	/usr/lib/go-1.17/src/fmt/print.go:709 +0x693 fp=0xc020360468 sp=0xc0203603c8 pc=0x4ad033
fmt.(*pp).doPrintf(0xc0052abe10, {0x5a01aa, 0x4}, {0xc0203605f8, 0x0, 0x0})
	/usr/lib/go-1.17/src/fmt/print.go:1026 +0x288 fp=0xc020360560 sp=0xc020360468 pc=0x4af8e8
fmt.Fprintf({0x664f40, 0xc00000e018}, {0x5a01aa, 0x4}, {0xc0203605f8, 0x1, 0x1})
	/usr/lib/go-1.17/src/fmt/print.go:204 +0x75 fp=0xc0203605c0 sp=0xc020360560 pc=0x4a9cb5
fmt.Printf(...)
	/usr/lib/go-1.17/src/fmt/print.go:213
github.com/gopcua/opcua/ua.(*Variant).ArrayLength(0xc0002ff240)
	/home/aleksandr/go/pkg/mod/github.com/gopcua/opcua@v0.2.2/ua/variant.go:104 +0x5d fp=0xc020360618 sp=0xc0203605c0 pc=0x521cfd
github.com/gopcua/opcua/ua.(*Variant).String(0xc0002ff240)
	/home/aleksandr/go/pkg/mod/github.com/gopcua/opcua@v0.2.2/ua/variant.go:496 +0x25 fp=0xc020360640 sp=0xc020360618 pc=0x524985
fmt.(*pp).handleMethods(0xc0052abd40, 0x0)
	/usr/lib/go-1.17/src/fmt/print.go:626 +0x383 fp=0xc0203608f0 sp=0xc020360640 pc=0x4ac483
fmt.(*pp).printArg(0xc0052abd40, {0x59ad00, 0xc0002ff240}, 0x76)
	/usr/lib/go-1.17/src/fmt/print.go:709 +0x693 fp=0xc020360990 sp=0xc0203608f0 pc=0x4ad033
fmt.(*pp).doPrintf(0xc0052abd40, {0x5a01aa, 0x4}, {0xc020360b20, 0x0, 0x0})
	/usr/lib/go-1.17/src/fmt/print.go:1026 +0x288 fp=0xc020360a88 sp=0xc020360990 pc=0x4af8e8
fmt.Fprintf({0x664f40, 0xc00000e018}, {0x5a01aa, 0x4}, {0xc020360b20, 0x1, 0x1})
	/usr/lib/go-1.17/src/fmt/print.go:204 +0x75 fp=0xc020360ae8 sp=0xc020360a88 pc=0x4a9cb5
fmt.Printf(...)
	/usr/lib/go-1.17/src/fmt/print.go:213
github.com/gopcua/opcua/ua.(*Variant).ArrayLength(0xc0002ff240)
	/home/aleksandr/go/pkg/mod/github.com/gopcua/opcua@v0.2.2/ua/variant.go:104 +0x5d fp=0xc020360b40 sp=0xc020360ae8 pc=0x521cfd
github.com/gopcua/opcua/ua.(*Variant).String(0xc0002ff240)
	/home/aleksandr/go/pkg/mod/github.com/gopcua/opcua@v0.2.2/ua/variant.go:496 +0x25 fp=0xc020360b68 sp=0xc020360b40 pc=0x524985
fmt.(*pp).handleMethods(0xc0052abc70, 0x0)
	/usr/lib/go-1.17/src/fmt/print.go:626 +0x383 fp=0xc020360e18 sp=0xc020360b68 pc=0x4ac483
fmt.(*pp).printArg(0xc0052abc70, {0x59ad00, 0xc0002ff240}, 0x76)
[... 208 lines cut ...]
	/usr/lib/go-1.17/src/testing/testing.go:1598 +0x6e
testing.tRunner(0xc0001e0340, 0xc00004fd18)
	/usr/lib/go-1.17/src/testing/testing.go:1259 +0x102
testing.runTests(0xc00015e380, {0x78c320, 0x47, 0x47}, {0x47734d, 0x5a2ebc, 0x78f0c0})
	/usr/lib/go-1.17/src/testing/testing.go:1596 +0x43f
testing.(*M).Run(0xc00015e380)
	/usr/lib/go-1.17/src/testing/testing.go:1504 +0x51d
main.main()
	_testmain.go:185 +0x14b

goroutine 534 [chan receive]:
testing.(*T).Run(0xc000324680, {0x5a2832, 0x1bd244f13564}, 0x634dd8)
	/usr/lib/go-1.17/src/testing/testing.go:1307 +0x375
github.com/gopcua/opcua/ua.TestArray(0xc0003244e0)
	/home/aleksandr/go/pkg/mod/github.com/gopcua/opcua@v0.2.2/ua/variant_test.go:508 +0x35
testing.tRunner(0xc000324680, 0x634e18)
	/usr/lib/go-1.17/src/testing/testing.go:1259 +0x102
created by testing.(*T).Run
	/usr/lib/go-1.17/src/testing/testing.go:1306 +0x35a
FAIL	github.com/gopcua/opcua/ua	4.464s
ok  	github.com/gopcua/opcua/uacp	(cached)
ok  	github.com/gopcua/opcua/uapolicy	(cached)
ok  	github.com/gopcua/opcua/uasc	(cached)
FAIL
`

fenogentov, 2021-11-22:
/home/aleksandr/go/pkg/mod/github.com/gopcua/opcua@v0.2.2/examples/browse
stopped working. Some other examples too

kung-foo, 2021-11-22:
The tests run fine on 1.17.3 on Linux (x86_64). What OS and platform are you on?

fenogentov, 2021-11-22:
Ubuntu 18.04.6 LTS (64)

kung-foo, 2021-11-22:
It seems like there must be something wrong with either your local copy of the repo, or possibly your go toolchain.

Here you can see the simplest "clean" checkout and test:

```
➜  ~ docker run --rm -it --entrypoint=bash -w /opt golang:1.17.3 
root@534366c551c8:/opt# git clone -q https://github.com/gopcua/opcua . 
root@534366c551c8:/opt# go test ./...
go: downloading github.com/pkg/errors v0.8.1
go: downloading github.com/pascaldekloe/goe v0.1.0
go: downloading golang.org/x/crypto v0.0.0-20200622213623-75b288015ac9
go: downloading golang.org/x/sys v0.0.0-20210809222454-d867a43fc93e
ok  	github.com/gopcua/opcua	0.012s
?   	github.com/gopcua/opcua/cmd/id	[no test files]
?   	github.com/gopcua/opcua/cmd/service	[no test files]
?   	github.com/gopcua/opcua/cmd/service/goname	[no test files]
?   	github.com/gopcua/opcua/cmd/status	[no test files]
?   	github.com/gopcua/opcua/debug	[no test files]
ok  	github.com/gopcua/opcua/errors	0.001s
?   	github.com/gopcua/opcua/examples/accesslevel	[no test files]
?   	github.com/gopcua/opcua/examples/browse	[no test files]
?   	github.com/gopcua/opcua/examples/crypto	[no test files]
?   	github.com/gopcua/opcua/examples/datetime	[no test files]
?   	github.com/gopcua/opcua/examples/endpoints	[no test files]
?   	github.com/gopcua/opcua/examples/history-read	[no test files]
?   	github.com/gopcua/opcua/examples/method	[no test files]
?   	github.com/gopcua/opcua/examples/monitor	[no test files]
?   	github.com/gopcua/opcua/examples/read	[no test files]
?   	github.com/gopcua/opcua/examples/regread	[no test files]
?   	github.com/gopcua/opcua/examples/server	[no test files]
?   	github.com/gopcua/opcua/examples/subscribe	[no test files]
?   	github.com/gopcua/opcua/examples/translate	[no test files]
?   	github.com/gopcua/opcua/examples/trigger	[no test files]
?   	github.com/gopcua/opcua/examples/udt	[no test files]
?   	github.com/gopcua/opcua/examples/write	[no test files]
?   	github.com/gopcua/opcua/id	[no test files]
?   	github.com/gopcua/opcua/monitor	[no test files]
ok  	github.com/gopcua/opcua/ua	0.042s
ok  	github.com/gopcua/opcua/uacp	2.038s
ok  	github.com/gopcua/opcua/uapolicy	0.768s
ok  	github.com/gopcua/opcua/uasc	0.003s
root@534366c551c8:/opt# 
```

fenogentov, 2021-11-30:
I have not been able to identify the problem.
Reinstalled Golang and libraries.
I use OPS UA servers: Prosys, Unified Automation CPP, Kepware, Siemens S1500
Reading tag data - all servers are running.
GetEndpoint - all work
Browse (ns=0;i=84) - everything except Prosys (opcua: browse children: EOF)

magiconair, 2021-12-06:
Can you try v0.2.3, please?

magiconair, 2021-12-06:
also, which go version do you use?

fenogentov, 2021-12-06:
go 17.3