gopcua/opcua #511: An invalid node ID error is reported after the read.go command is executed

wangpengzong, 2021-11-24:
Kepware
```
V6.9.572.0
```
[image]
[image]
The Softing OPC UA Client reads data correctly
[image]

However, an invalid node ID error was reported. 
```
>go run examples/read/read.go -endpoint opc.tcp://localhost:49320 -node 'ns=0;i=2262'

>invalid node id: opcua: invalid node id: 'ns=0;i=2262'
exit status 1

>go run examples/read/read.go -endpoint opc.tcp://localhost:49320 -node ' ns=0;i=2262 '

>Status not OK: The node id refers to a node that does not exist in the server address space. StatusBadNodeIDUnknown (0x80340000)，TimestampsToReturnBoth: TimestampsToReturnBoth
exit status 1
```

magiconair, 2021-11-24:
Try reading just `i=2262`

kung-foo, 2021-11-24:
Also in your second invocation there are spaces before and after the node id. The node id parser is pretty strict, so probably won't handle that.

magiconair, 2021-11-24:
`ns=0;` shouldn't matter...

https://github.com/gopcua/opcua/blob/main/ua/node_id_test.go#L143

Also, good point about the spaces. @kung-foo we should add test cases for this. Do you think it is worth to trim spaces in `ParseNodeID` ?

wangpengzong, 2021-11-24:
Using ```i=2262``` is still an error,
```
>go run examples/read/read.go -endpoint opc.tcp://localhost:49320 -node 'i=2262'

>Status not OK: The node id refers to a node that does not exist in the server address space. StatusBadNodeIDUnknown (0x80340000)，TimestampsToReturnBoth: TimestampsToReturnBoth
exit status 1
```
[image]

kung-foo, 2021-11-24:
Are you able to read `i=2258` (current time)?

wangpengzong, 2021-11-25:
> Are you able to read `i=2258` (current time)?

Reading ```i=2258``` is the same error message, but I can get the time using ```datetime.go```

```
>go run examples/read/read.go -endpoint opc.tcp://localhost:49320 -node 'i=2258'

>Status not OK: The node id refers to a node that does not exist in the server address space. StatusBadNodeIDUnknown (0x80340000)，TimestampsToReturnBoth: TimestampsToReturnBoth
exit status 1

>go run examples/datetime/datetime.go -endpoint opc.tcp://localhost:49320

>* http://opcfoundation.org/UA/SecurityPolicy#None MessageSecurityModeNone
2021-11-25 00:31:01.1735585 +0000 UTC
```

wangpengzong, 2021-11-25:
# ```read.go```
并且，在读取 ```Object/Server``` 目录下的Node Id时，在KepServer中将会输出 ```ActivateSessionRequest,ActivateSessionResponse,ReadRequest,ReadResponse``` 一共四条日志，但是在读取 ```Object/channel1/equip1``` 目录下的Node Id时，在KepServer中只会输出 ```ActivateSessionRequest,ActivateSessionResponse``` 一共两条日志 
[image]

# The ```Object/Server``` directory reads logs
```ActivateSessionRequest```
[image]
```ActivateSessionResponse```
[image]
```ReadRequest```
[image]
```ReadResponse```
[image]

# The ```Object/channel1/equip1``` directory reads logs
```ActivateSessionRequest```
[image]
```ActivateSessionResponse```
[image]

  # ```datetime.go```
Six records were generated when read using ```datetime.go```,They are respectively ```ActivateSessionRequest, ActivateSessionResponse, ReadRequest ReadResponse, CloseSessionRequest, CloseSessionResponse```
```ActivateSessionRequest```
[image]
```ActivateSessionResponse```
[image]
```ReadRequest```
[image]
```ReadResponse```
[image]
```CloseSessionRequest```
[image]
```CloseSessionResponse```
[image]

magiconair, 2021-11-25:
In the read.go example the dump shows `s='i=2258'` which means it is trying to read the node as a string node id which cannot work of course. 

Did you modify the examples in any way?

magiconair, 2021-11-25:
[image]

magiconair, 2021-11-25:
Terminal 1:
```sh
> ./uatest/method_server.py
cryptography is not installed, use of crypto disabled
cryptography is not installed, use of crypto disabled
Endpoints other than open requested but private key and certificate are not set.
Listening on 0.0.0.0:4840
```

Terminal 2:
```sh
> git describe
v0.2.2
> go run ./examples/read/read.go  -node 'i=2258'
time.Date(2021, time.November, 25, 7, 39, 29, 558650000, time.UTC)
```

wangpengzong, 2021-11-26:
Now the data can be obtained through the command box of ```git```, why can't it be obtained by using the ```cmd``` command box of the windows system
[image]

magiconair, 2021-11-26:
Can you try to use `-node "i=2258"`? I think the windows shell does not remove the single quotes.

wangpengzong, 2021-11-26:
Yes, double quotes will work, thank you
```
>go run examples/read/read.go -endpoint opc.tcp://localhost:49320 -node "i=2262"

"urn:LAPTOP-B5F5AJI1:Kepware.KEPServerEX.V6:UA%20Server"
```

closed 2021-11-26 by wangpengzong