gopcua/opcua #531: connect to server with more than 200 nodeID will get EOF

whutwxn, 2021-12-20:
read nodeIDs for about 3000 will get EOF error;
here is the desprition:
debug: Connecting to opc.tcp://192.168.1.33:4840
debug: conn 1: start HEL/ACK handshake
debug: conn 1: sent HELF with 59 bytes
debug: conn 1: recv ACKF with 28 bytes
debug: conn 1: server has no chunk limit. Using 512
debug: conn 1: server has no message size limit. Using 2097152
debug: conn 1: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0x2000, SendBufSize:0x2000, MaxMessageSize:0x200000, MaxChunkCount:0x200}
debug: uasc 1/1: send *ua.OpenSecureChannelRequest with 132 bytes
debug: conn 1: recv OPNF with 136 bytes
debug: uasc OPN Request
debug: uasc 1/1: recv OPNF with 136 bytes
debug: uasc 1/1: recv *ua.OpenSecureChannelResponse
debug: sending *ua.OpenSecureChannelResponse to handler
debug: received security token: channelID=2721149132 tokenID=1 createdAt=2021-12-20T01:32:43Z lifetime=1h0m0s
debug: channelID 2721149132 will be refreshed in 45m0s (2021-12-20T02:17:38Z)
debug: uasc 1/2: send *ua.GetEndpointsRequest with 96 bytes
debug: channelID 2721149132/1 will expire at 2021-12-20T02:47:43Z
debug: conn 1: recv MSGF with 4231 bytes
debug: uasc 1/2: recv MSGF with 4231 bytes
debug: uasc 1/2: recv *ua.GetEndpointsResponse
debug: sending *ua.GetEndpointsResponse to handler
opc ua close error: <nil>
debug: uasc Close()
debug: uasc 1/3: send *ua.CloseSecureChannelRequest with 57 bytes
debug: conn 1: close
debug: uasc readChunk EOF
debug: Loading cert/key from cert.pem/key.pem
Using config:
Endpoint: opc.tcp://192.168.1.33:4840
Security mode: http://opcfoundation.org/UA/SecurityPolicy#Basic256, MessageSecurityModeSignAndEncrypt
Auth mode : UserTokenTypeUserName
debug: Connecting to opc.tcp://192.168.1.33:4840
debug: conn 2: start HEL/ACK handshake
debug: conn 2: sent HELF with 59 bytes
debug: conn 2: recv ACKF with 28 bytes
debug: conn 2: server has no chunk limit. Using 512
debug: conn 2: server has no message size limit. Using 2097152
debug: conn 2: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0x2000, SendBufSize:0x2000, MaxMessageSize:0x200000, MaxChunkCount:0x200}
debug: uasc 2/1: send *ua.OpenSecureChannelRequest with 1427 bytes
debug: conn 2: recv OPNF with 1612 bytes
debug: uasc OPN Request
debug: Setting securityPolicy to http://opcfoundation.org/UA/SecurityPolicy#Basic256
debug: uasc 2/1: recv OPNF with 1612 bytes
debug: uasc 2/1: recv *ua.OpenSecureChannelResponse
debug: sending *ua.OpenSecureChannelResponse to handler
debug: received security token: channelID=2721149133 tokenID=1 createdAt=2021-12-20T01:32:44Z lifetime=1h0m0s
debug: channelID 2721149133 will be refreshed in 45m0s (2021-12-20T02:17:40Z)
debug: channelID 2721149133/1 will expire at 2021-12-20T02:47:44Z
debug: uasc 2/2: send *ua.CreateSessionRequest with 1120 bytes
debug: conn 2: recv MSGF with 5648 bytes
debug: uasc 2/2: recv MSGF with 5648 bytes
debug: uasc 2/2: recv *ua.CreateSessionResponse
debug: sending *ua.CreateSessionResponse to handler
debug: uasc 2/3: send *ua.ActivateSessionRequest with 768 bytes
debug: conn 2: recv MSGF with 128 bytes
debug: uasc 2/3: recv MSGF with 128 bytes
debug: uasc 2/3: recv *ua.ActivateSessionResponse
debug: sending *ua.ActivateSessionResponse to handler
debug: client: monitor: start
debug: sub: pause
debug: uasc 2/4: send *ua.ReadRequest with 128 bytes
debug: conn 2: recv MSGF with 112 bytes
debug: uasc 2/4: recv MSGF with 112 bytes
debug: uasc 2/4: recv *ua.ReadResponse
debug: sending *ua.ReadResponse to handler
Server's Time | Conn 1 2021-12-20 01:32:47.3424632 +0000 UTC |
debug: uasc 2/5: send *ua.ReadRequest with 225456 bytes
debug: uasc readChunk EOF
Read failed: EOF

siscia, 2022-01-03:
Try to send smaller requests.

I am thinking that maybe the OPC/UA server dropped the connection because it would take too much time.

This is really depending on the OPC/UA server implementation and setting though.

Mebus, 2022-01-04:
Hallo,

unfortunately I also only get a "EOF" when I try to connect to a Beckhoff PLC using this example:

https://github.com/gopcua/opcua/blob/main/examples/crypto/crypto.go

This is the output:

```
$ go run . -endpoint opc.tcp://192.168.0.10:4840 -sec-policy Basic256Sha256 -sec-mode SignAndEncrypt
secMode:  MessageSecurityModeSignAndEncrypt secPolicy: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
Using config:
Endpoint: opc.tcp://192.168.0.10:4840
Security mode: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256, MessageSecurityModeSignAndEncrypt
Auth mode : UserTokenTypeAnonymous
EOF
exit status 1
```

Why is this not working? How can I send smaller requests?

Greetings 

Mebus

Mebus, 2022-01-04:
Hallo siscia,

I also tried a public OPC/UA server on the internet and I get a similar result:

```
$ go run . -endpoint opc.tcp://opcuademo.sterfive.com:26543 -debug
debug: uacp: connecting to opc.tcp://opcuademo.sterfive.com:26543
debug: uacp 1: start HEL/ACK handshake
debug: uacp 1: sent HELF with 70 bytes
debug: uacp 1: recv ACKF with 28 bytes
debug: uacp 1: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x4000000, MaxChunkCount:0xffff}
debug: uasc 1/1: send *ua.OpenSecureChannelRequest with 132 bytes
debug: uacp 1: recv OPNF with 135 bytes
debug: uasc 1/1: recv OPNF with 135 bytes
debug: uasc 1/1: recv *ua.OpenSecureChannelResponse
debug: uasc 1/1: sending *ua.OpenSecureChannelResponse to handler
debug: uasc 1: received security token. channelID=1013739 tokenID=1 createdAt=2022-01-04T16:28:24Z lifetime=10m0s
debug: uasc 1: security token is refreshed at 2022-01-04T16:35:53Z (7m30s). channelID=1013739 tokenID=1
debug: uasc 1/2: send *ua.GetEndpointsRequest with 107 bytes
debug: uasc 1: security token expires at 2022-01-04T16:40:54Z. channelID=1013739 tokenID=1
debug: uacp 1: recv MSGF with 15137 bytes
debug: uasc 1/2: recv MSGF with 15137 bytes
debug: uasc 1/2: recv *ua.GetEndpointsResponse
debug: uasc 1/2: sending *ua.GetEndpointsResponse to handler
debug: uasc 1: Close()
debug: uasc 1/3: send *ua.CloseSecureChannelRequest with 57 bytes
debug: uacp 1: close
debug: uasc 1: readChunk EOF
debug: Loading cert/key from cert.pem/key.pem
Using config:
Endpoint: opc.tcp://opcuademo.sterfive.com:26543
Security mode: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256, MessageSecurityModeSignAndEncrypt
Auth mode : UserTokenTypeAnonymous
debug: uacp: connecting to opc.tcp://opcuademo.sterfive.com:26543
debug: uacp 2: start HEL/ACK handshake
debug: uacp 2: sent HELF with 70 bytes
debug: uacp 2: recv ACKF with 28 bytes
debug: uacp 2: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x4000000, MaxChunkCount:0xffff}
debug: uasc 2/1: send *ua.OpenSecureChannelRequest with 1437 bytes
debug: uacp 2: recv ERRF with 53 bytes
debug: uasc 2: readChunk EOF
debug: uasc 2: failed to open a new secure channel
debug: uacp 2: close
EOF
exit status 1
$ go version
go version go1.17.5 windows/amd64
```

Would be great if you could help.

Thanks

Mebus

magiconair, 2022-01-04:
I can have a look later. Which version are you using?

—
Frank Schröder

siscia, 2022-01-04:
In my experience `EOF` means that the PLC or the OPC/UA server drop the connection.

So the go code tries to follow the OPCUA protocol and read let's say 100 bytes, because that many bytes are expected from the protocol, but it cannot read the whole amount of bytes and so it returns an `EOF`. 

The reason why this happen is not always clear and not always the same!

In the original question, @whutwxn , was trying to send a very big request 200k bytes, and it might be that the server refuse to reply to the client. Again, it is just speculation.

You can see it here:

```
 debug: uasc 2/5: send *ua.ReadRequest with 225456 bytes
 ```

In the case of @Mebus it seems to be something different!

In both cases, it fails around setting up a secure channel:

```
debug: conn 2: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x3fffc0, MaxChunkCount:0x40}
debug: uasc 2/1: send *ua.OpenSecureChannelRequest with 956 bytes
debug: conn 2: recv ERRF with 16 bytes
```

@magiconair is much better equipped than me to help with this discussion.

My suggestion for @whutwxn would it be to try making a smaller request and see if it works or not.

For @Mebus it will be to try with the latest tagged releases, I suggest starting from version `0.2.1` and moving up to the latest one. 
For each version that does not work I suggest posting the code and the debug trace.

Please note that this is an open source project.

Mebus, 2022-01-05:
Hallo siscia,

I know that it is a open source project and I really appreciate your work. I was using open62541 before. However I discovered that Go is much quicker to program than C/C++ 😃 . So thank you for this Go implementation.

Here are some of my results of the tests with a Beckhoff PLC.

I am using the crypto example directly in the repo.

This is the result on version 0.2.1:

```
go run . -endpoint opc.tcp://192.168.14.45:4840 -sec-policy None -sec-mode None -auth-mode UserName -debug
debug: Connecting to opc.tcp://192.168.14.45:4840
debug: conn 1: start HEL/ACK handshake
debug: conn 1: sent HELF with 60 bytes
debug: conn 1: recv ACKF with 28 bytes
debug: conn 1: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x3fffc0, MaxChunkCount:0x40}
debug: uasc 1/1: send *ua.OpenSecureChannelRequest with 132 bytes
debug: conn 1: recv OPNF with 136 bytes
debug: uasc OPN Request
debug: uasc 1/1: recv OPNF with 136 bytes
debug: uasc 1/1: recv *ua.OpenSecureChannelResponse
debug: sending *ua.OpenSecureChannelResponse to handler
debug: received security token: channelID=323908194 tokenID=1 createdAt=2022-01-05T08:04:25Z lifetime=1h0m0s
debug: channelID 323908194 will be refreshed in 45m0s (2022-01-05T08:49:23Z)
debug: uasc 1/2: send *ua.GetEndpointsRequest with 97 bytes
debug: channelID 323908194/1 will expire at 2022-01-05T09:19:25Z
debug: conn 1: recv MSGF with 17975 bytes
debug: uasc 1/2: recv MSGF with 17975 bytes
debug: uasc 1/2: recv *ua.GetEndpointsResponse
debug: sending *ua.GetEndpointsResponse to handler
debug: uasc Close()
debug: uasc 1/3: send *ua.CloseSecureChannelRequest with 57 bytes
debug: conn 1: close
debug: uasc readChunk EOF
debug: Loading cert/key from cert.pem/key.pem
Failed to load certificate: open cert.pem: The system cannot find the file specified.
Enter username: Admin
Enter password:
secMode:  MessageSecurityModeNone secPolicy: http://opcfoundation.org/UA/SecurityPolicy#None
Using config:
Endpoint: opc.tcp://MYMACHINE:4840
Security mode: http://opcfoundation.org/UA/SecurityPolicy#None, MessageSecurityModeNone
Auth mode : UserTokenTypeUserName
debug: Connecting to opc.tcp://192.168.14.45:4840
debug: conn 2: start HEL/ACK handshake
debug: conn 2: sent HELF with 60 bytes
debug: conn 2: recv ACKF with 28 bytes
debug: conn 2: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x3fffc0, MaxChunkCount:0x40}
debug: uasc 2/1: send *ua.OpenSecureChannelRequest with 132 bytes
debug: conn 2: recv OPNF with 136 bytes
debug: uasc OPN Request
debug: uasc 2/1: recv OPNF with 136 bytes
debug: uasc 2/1: recv *ua.OpenSecureChannelResponse
debug: sending *ua.OpenSecureChannelResponse to handler
debug: received security token: channelID=323908197 tokenID=1 createdAt=2022-01-05T08:04:31Z lifetime=1h0m0s
debug: channelID 323908197 will be refreshed in 45m0s (2022-01-05T08:49:29Z)
debug: uasc 2/2: send *ua.CreateSessionRequest with 267 bytes
debug: channelID 323908197/1 will expire at 2022-01-05T09:19:31Z
debug: conn 2: recv MSGF with 20034 bytes
debug: uasc 2/2: recv MSGF with 20034 bytes
debug: uasc 2/2: recv *ua.CreateSessionResponse
debug: sending *ua.CreateSessionResponse to handler
debug: uasc 2/3: send *ua.ActivateSessionRequest with 701 bytes
debug: conn 2: recv MSGF with 52 bytes
debug: uasc 2/3: recv MSGF with 52 bytes
debug: uasc 2/3: err: User does not have permission to perform the requested operation. StatusBadUserAccessDenied (0x801F0000)
debug: sending *ua.ServiceFault to handler
debug: uasc Close()
debug: uasc 2/4: send *ua.CloseSecureChannelRequest with 57 bytes
[... 240 lines cut ...]
secMode:  MessageSecurityModeSignAndEncrypt secPolicy: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
Using config:
Endpoint: opc.tcp://MYMACHINE:4840
Security mode: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256, MessageSecurityModeSignAndEncrypt
Auth mode : UserTokenTypeUserName
debug: Connecting to opc.tcp://192.168.14.45:4840
debug: conn 2: start HEL/ACK handshake
debug: conn 2: sent HELF with 60 bytes
debug: conn 2: recv ACKF with 28 bytes
debug: conn 2: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x3fffc0, MaxChunkCount:0x40}
debug: uasc 2/1: send *ua.OpenSecureChannelRequest with 1437 bytes
debug: conn 2: recv ERRF with 16 bytes
debug: uasc readChunk EOF
debug: failed to open a new secure channel
EOF
exit status 1
```

I discovered that when the `cert.pem` and `key.pem` files have been generated and are present in the crypto example it stops working for version 0.2.1. When I delete them it works again or at least it says that the user doesn't have the permissions instead of just returning a EOF. As soon as I run it with the `-gen-cert` option so that the files are present again, it stops working.

Can you identify some causes of this?

Thanks!

Mebus

magiconair, 2022-01-05:
Hi @Mebus testing with the latest version is good enough. We're at `v0.2.6` now. Note that `v0.2.4` and `v0.2.5` are broken. Don't use those.

Here is my read test against the public server with `v0.2.6`

```
go run examples/read/read.go -endpoint opc.tcp://opcuademo.sterfive.com:26543 -debug -node 'i=2258'
debug: uacp: connecting to opc.tcp://opcuademo.sterfive.com:26543
debug: uacp 1: start HEL/ACK handshake
debug: uacp 1: sent HELF with 70 bytes
debug: uacp 1: recv ACKF with 28 bytes
debug: uacp 1: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x4000000, MaxChunkCount:0xffff}
debug: uasc 1/1: send *ua.OpenSecureChannelRequest with 132 bytes
debug: uacp 1: recv OPNF with 135 bytes
debug: uasc 1/1: recv OPNF with 135 bytes
debug: uasc 1/1: recv *ua.OpenSecureChannelResponse
debug: uasc 1/1: sending *ua.OpenSecureChannelResponse to handler
debug: uasc 1: received security token. channelID=1047261 tokenID=1 createdAt=2022-01-05T08:26:58Z lifetime=10m0s
debug: uasc 1: security token is refreshed at 2022-01-05T08:34:28Z (7m30s). channelID=1047261 tokenID=1
debug: uasc 1: security token expires at 2022-01-05T08:39:28Z. channelID=1047261 tokenID=1
debug: uasc 1/2: send *ua.CreateSessionRequest with 277 bytes
debug: uacp 1: recv MSGF with 5304 bytes
debug: uasc 1/2: recv MSGF with 5304 bytes
debug: uasc 1/2: recv *ua.CreateSessionResponse
debug: uasc 1/2: sending *ua.CreateSessionResponse to handler
debug: uasc 1/3: send *ua.ActivateSessionRequest with 133 bytes
debug: uacp 1: recv MSGF with 96 bytes
debug: uasc 1/3: recv MSGF with 96 bytes
debug: uasc 1/3: recv *ua.ActivateSessionResponse
debug: uasc 1/3: sending *ua.ActivateSessionResponse to handler
debug: sub: pause
debug: client: monitor: start
debug: uasc 1/4: send *ua.ReadRequest with 115 bytes
debug: uacp 1: recv MSGF with 499 bytes
debug: uasc 1/4: recv MSGF with 499 bytes
debug: uasc 1/4: recv *ua.ReadResponse
debug: uasc 1/4: sending *ua.ReadResponse to handler
debug: uasc 1/5: send *ua.ReadRequest with 112 bytes
debug: uacp 1: recv MSGF with 90 bytes
debug: uasc 1/5: recv MSGF with 90 bytes
debug: uasc 1/5: recv *ua.ReadResponse
debug: uasc 1/5: sending *ua.ReadResponse to handler
time.Date(2022, time.January, 5, 8, 27, 1, 119000000, time.UTC)
debug: uasc 1/6: send *ua.CloseSessionRequest with 79 bytes
debug: uacp 1: recv MSGF with 52 bytes
debug: uasc 1/6: recv MSGF with 52 bytes
debug: uasc 1/6: recv *ua.CloseSessionResponse
debug: uasc 1/6: sending *ua.CloseSessionResponse to handler
debug: uasc 1: Close()
debug: client: monitor: done
debug: sub: pause: ctx.Done()
debug: sub: done
debug: uasc 1/7: send *ua.CloseSecureChannelRequest with 57 bytes
debug: uacp 1: close
debug: uasc 1: readChunk EOF
```

magiconair, 2022-01-05:
Unencrypted communication definitely works with all versions. If not, please report.

magiconair, 2022-01-05:
@Mebus could you create a `tcpdump`?

magiconair, 2022-01-05:
@whutwxn your `MaxMessageSize` is 0x20000 which is 131072 bytes but you are sending a request with 220000 bytes. Try increasing the `MaxMessageSize`.

magiconair, 2022-01-05:
@Mebus maybe try lowering the `MaxMessageSize`. It is currently at `0x3fffc0`

Mebus, 2022-01-05:
> @Mebus maybe try lowering the `MaxMessageSize`. It is currently at `0x3fffc0`

How can I do that?

magiconair, 2022-01-05:
Use the `opcua.MaxMessageSize` option like this:

```go
	c := opcua.NewClient(*endpoint,
		opcua.SecurityMode(ua.MessageSecurityModeNone),
		opcua.MaxMessageSize(0x1000000),
	)
```

Mebus, 2022-01-05:
Hi magiconair,

I tried it like this:

https://gist.github.com/Mebus/25bf352722f97af8dc385207fa00f6f7

See lines 62 and 63. And I got this:

```
go run .\examples\crypto -endpoint opc.tcp://192.168.14.45:4840 -sec-policy None -sec-mode None -auth-mode UserName -debug
debug: uacp: connecting to opc.tcp://192.168.14.45:4840
debug: uacp 1: start HEL/ACK handshake
debug: uacp 1: sent HELF with 60 bytes
debug: uacp 1: recv ACKF with 28 bytes
debug: uacp 1: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x3fffc0, MaxChunkCount:0x40}
debug: uasc 1/1: send *ua.OpenSecureChannelRequest with 132 bytes
debug: uacp 1: recv OPNF with 136 bytes
debug: uasc 1/1: recv OPNF with 136 bytes
debug: uasc 1/1: recv *ua.OpenSecureChannelResponse
debug: uasc 1/1: sending *ua.OpenSecureChannelResponse to handler
debug: uasc 1: received security token. channelID=323909287 tokenID=1 createdAt=2022-01-05T09:28:30Z lifetime=1h0m0s
debug: uasc 1/2: send *ua.GetEndpointsRequest with 97 bytes
debug: uasc 1: security token is refreshed at 2022-01-05T10:13:29Z (45m0s). channelID=323909287 tokenID=1
debug: uasc 1: security token expires at 2022-01-05T10:43:30Z. channelID=323909287 tokenID=1
debug: uacp 1: recv MSGF with 17975 bytes
debug: uasc 1/2: recv MSGF with 17975 bytes
debug: uasc 1/2: recv *ua.GetEndpointsResponse
debug: uasc 1/2: sending *ua.GetEndpointsResponse to handler
debug: uasc 1: Close()
debug: uasc 1/3: send *ua.CloseSecureChannelRequest with 57 bytes
debug: uacp 1: close
debug: uasc 1: readChunk EOF
debug: Loading cert/key from cert.pem/key.pem
Enter username: Admin
Enter password:
secMode:  MessageSecurityModeNone secPolicy: http://opcfoundation.org/UA/SecurityPolicy#None
Using config:
Endpoint: opc.tcp://MYMACHINE:4840
Security mode: http://opcfoundation.org/UA/SecurityPolicy#None, MessageSecurityModeNone
Auth mode : UserTokenTypeUserName
debug: uacp: connecting to opc.tcp://192.168.14.45:4840
debug: uacp 2: start HEL/ACK handshake
debug: uacp 2: sent HELF with 60 bytes
debug: uacp 2: recv ACKF with 28 bytes
debug: uacp 2: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xf83e, MaxMessageSize:0x3fffc0, MaxChunkCount:0x40}
debug: uasc 2/1: send *ua.OpenSecureChannelRequest with 956 bytes
debug: uacp 2: recv ERRF with 16 bytes
debug: uasc 2: readChunk EOF
debug: uasc 2: failed to open a new secure channel
debug: uacp 2: close
EOF
exit status 1
```

Did I do that correctly? It still sys "MaxMessageSize:0x3fffc0".

Mebus

Mebus, 2022-01-05:
> @Mebus could you create a `tcpdump`?

May I send it to you privately, because I don't want to disclose any information publicly?

Mebus

magiconair, 2022-01-05:
@Mebus sure. You can find me on Keybase.

Mebus, 2022-01-05:
> @Mebus sure. You can find me on Keybase.

I will send it to you later.

Mebus, 2022-01-06:
> > @Mebus sure. You can find me on Keybase.
> 
> I will send it to you later.

Done! 😃

magiconair, 2022-01-06:
Thank you. It's a bank holiday here today. I'll look at it tomorrow.

Mebus, 2022-01-06:
Hallo folks,

I looked into the tcpdumps myself and compared it with UAExpert:

[image]

There seems to be a difference in the ClientNonce during the opening of the SecureChannel, which is then aborted by the server.

The ClientNonce seems to have an influence on the connection establishment, at least I found a similiar issue in python-opcua:

https://github.com/FreeOpcUa/python-opcua/issues/393

Can I somehow set the ClientNonce?

Mebus

Mebus, 2022-01-06:
Hm. I set it to "00" in the Go client, but it did not help.

Mebus, 2022-01-06:
Hi,

I have also created a log on the Beckhoff-Server side. It is a snipped from when I ran the command

```
 go run .\examples\crypto\ -endpoint opc.tcp://192.168.14.45:4840 -sec-policy Basic256Sha256 -sec-mode SignAndEncrypt -auth-mode UserName -debug
````

[beckhofflog.txt](https://github.com/gopcua/opcua/files/7822054/beckhofflog.txt)

On the server side the certificate is always beeing rejected, even if I put it into the "trusted" folder.

Mebus

magiconair, 2022-01-06:
Does the server trust the signer?

Mebus, 2022-01-06:
Yes, I think so. I put it into the "trusted" directory. I did nothing else with the UAExpert certificates and they work.

Mebus

Mebus, 2022-01-10:
@magiconair Have you already been able to look into the log files I have sent you?

magiconair, 2022-01-10:
Not yet. Maybe today. Otherwise tomorrow. Still school holidays in Sweden

Mebus, 2022-01-24:
@magiconair Have you been able to look at it? I also tried it with keys that worked with open62541, but that also didn't help :-(

magiconair, 2022-01-25:
let me have a look now. I was down with Covid for a week.

magiconair, 2022-01-25:
Replied via email. Can you find me on keybase?

magiconair, 2022-01-27:
@Mebus BTW, there is nothing special or secret in these tcpdumps unless I'm missing something.

Mebus, 2022-01-27:
@magiconair Yes, but you never know. I'll continue testing next week. Unfortunately, time is a bit tight this week.

closed 2025-04-25 by github-actions[bot]