gopcua/opcua #547: Update namespaces array when reconnecting

cailloumajor, 2022-01-15:
Hello, and thank you for your work on this project.
I am writing a proxy to allow subscribing to OPC-UA nodes realtime updates via [Centrifugo](https://centrifugal.dev/): https://github.com/cailloumajor/opcua-centrifugo
I currently explore about how to use namespace URL, I read the code and also https://github.com/gopcua/opcua/pull/512. This got me to a question: shouldn't the client cached namespaces array be refreshed in the reconnection process ?
I would be happy to submit a PR if needed.

magiconair, 2022-01-27:
Yes, I think that's a good idea. PR would be welcome.

magiconair, 2022-01-27:
It was easy to add. @cailloumajor could you test if this works for you?

commit 0264c0e2 (2022-01-27): client: update namespaces on reconnect

Update the cached namespace array when reconnecting.

Fixes #547
PR #552 (2022-01-27): client: update namespaces on reconnect
Update the cached namespace array when reconnecting.

Fixes #547
closed 2022-01-27 by magiconair