gopcua/opcua #556: Reconnect client after server reboot

matthieu4294967296moineau, 2022-01-31:
What is the expected behavior of a client if server is rebooted (when I power cycle my PLC) ?

My client is connected to server via Ethernet cable, if I disconnect the cable and call `client.State()`, I still get `Connected` status. When I connect again the Ethernet cable, I get `Reconnecting` status. So checking for client state, I am able to reconnect the client to the server when connection is available. (even if getting `Connected` status after disconnecting the cable still seems weird to me)

However, if I unplug the power supply to PLC, and then call `client.Connect(ctx)` I get the error: 
`{"error": "dial tcp 192.168.4.102:4840: connect: connection refused"}`.
I am actually trying to reconnect in a loop every 2 seconds; after getting this error 4 or 5 times, I get this error: 
`panic: runtime error: invalid memory address or nil pointer dereference`

Is there a way to make the client reconnect after the reboot ?
Should I create another client after losing the connection ?

Thank you

myahuang, 2022-02-03:
On general,OPCUAserver collecting data from PLC  controllers, OPCuaclient poll data from OPCUAserver, client.State() indicate the communication status between client and server.If you want to find the status of communication between server and PLC,you can use method Node(args).Value(),it return data and err,if the err!=nil, the PLC is disconnected from the server.

myahuang, 2022-02-03:
This is a wonderful project,take my honor from China!

closed 2025-04-24 by github-actions[bot]