gopcua/opcua #560: How can I guarantee the monitor data order？

flykarry, 2022-02-24:
[image]
Is there any method to get the nodeId string  ?

tekulvw, 2022-02-25:
+1 on this, are we guaranteed that the monitored items in the data change notification matches the order in which items were added to monitor? Does this hold true across all number of monitor and unmonitor calls?

tekulvw, 2022-02-25:
After doing a bit of digging, it appears that the "clientHandle" you set when creating a new monitored item request is the intended method to link a MonitoredItem back to it's associated node:
[image]
[image]

flykarry, 2022-02-25:
> After doing a bit of digging, it appears that the "clientHandle" you set when creating a new monitored item request is the intended method to link a MonitoredItem back to it's associated node: [image] [image]

but in my example, I monitor 3 point with 1 clienthandle, the MonitoredItem is not sorted when I debug the code, in other words maybe I will receive not sorted array when the network is unstable

magiconair, 2022-02-25:
The clientHandle needs to be different for every monitored item. You use it to map back to the node. Check the subscription monitor. It should do something like this. The order is not guaranteed.

flykarry, 2022-02-25:
> The clientHandle needs to be different for every monitored item. You use it to map back to the node. Check the subscription monitor. It should do something like this. The order is not guaranteed.

Ok, It works. Thanks for your reply

closed 2022-03-05 by magiconair