gopcua/opcua #584: UserIdentityToken EncryptionAlgorithm RSA-OAEP-SHA1 uses non-existent URL

LarsStegman, 2022-06-02:
Hello,

I use the Telegraf plugin for OPC UA and want to connect to a OPC UA of a supplier of ours. Their server only support `http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep` as the SecurityPolicyUri for the UserTokenPolicy.  The supplier has used the [open62541](https://github.com/open62541/open62541) library.

When the gopcua library encrypts the ActivateSessionRequest user password, it indicates it uses the `http://opcfoundation.org/ua/security/rsa-oaep-sha1` algorithm to encrypt the password, whereas uaExpert claims to use `http://www.w3.org/2001/04/xmlenc#rsa-oaep`. 

It looks like this may be the cause of the problem, as we get a `0x80200000 [BadIdentityTokenInvalid]` response from the server. 

Sometimes the `w3.org` url is referenced in the encyrption algorithms, and sometimes the `opcfoundation.org`. I have compiled a version of Telegraf with the fix below, and the connection worked properly, including the password encryption. 

Will you accept a PR to with this change?

```diff
github.com\gopcua\opcua@v0.3.1\uapolicy\policyAes128Sha256RsaOaep.go
- encryptionURI:         "http://opcfoundation.org/ua/security/rsa-oaep-sha1",
+ encryptionURI:         "http://www.w3.org/2001/04/xmlenc#rsa-oaep",
```

magiconair, 2022-06-02:
This URL is probably there for a reason. Can we make it configurable?

LarsStegman, 2022-06-02:
Maybe. I find it strange that it does not resolve. I can't really find anything that looks like it either, to be honest.

LarsStegman, 2022-06-03:
I just noticed that the documentation mentions it should be the w3.org URI as well: https://github.com/gopcua/opcua/blob/278231bc74f43b9851f22ea99a32b57d610b7080/uapolicy/policyAes128Sha256RsaOaep.go#L40

dwhutchison, 2022-06-07:
I think the w3 address is correct as well.  I've searched and cannot find references to `http://opcfoundation.org/ua/security/rsa-oaep-sha1` in the spec so I'm not sure where that originally came from (pretty sure this is my fault >.<)

[Spec Part 7](https://www.opcfoundation.org/developer-tools/specifications-unified-architecture/part-7-profiles/) and [Aes128-Sha256-RsaOaep
](https://profiles.opcfoundation.org/profile/913) show the encryption URI should be `http://www.w3.org/2001/04/xmlenc#rsa-oaep`

LarsStegman, 2022-06-07:
There are several references to the opcfoundation URIs, I noticed. I haven't had time to check what their values should be, though.

PR #585 (2022-06-15): Change encryption URI for aes128Sha256RsaOaep to w3.org
Fix #584
closed 2022-06-15 by magiconair