gopcua/opcua #592: field UserIdentityTokens.SeucirtyPolicyURI is not filled - NC Simulator Heidenhain TC620

WinXaito, 2022-07-18:
Hello,

I have encountered a problem with a Heidenhain NC. (Tested with TNC620 simulator 817605-8, supporting OPC UA).

The value `UserTokenPolicy.SecurityPolicyURI` is not set. I don't know if this is a bug in Heidenhain, or a bug in this library. I don't know enough about the OPC UA standard to know.

Therefore, when using a `SignAndEncrypt` mode and certificate authentication, I have a crash on the following line (client.go:794), the value `s.cfg.AuthPolicyURI` is empty.

[image]

[image]

The following error is returned by the `crypto` example program:

```
error creating session signature: opcua: unsupported security policy opcua: unsupported security policy 
```

----------------------

To solve the error, I simply added the following code to the crypto.go example program here: https://github.com/gopcua/opcua/blob/main/examples/crypto/crypto.go#L225

```go
for _, u := range serverEndpoint.UserIdentityTokens {
	if u.SecurityPolicyURI == "" {
		u.SecurityPolicyURI = serverEndpoint.SecurityPolicyURI
	}
}
opts = append(opts, opcua.SecurityFromEndpoint(serverEndpoint, authMode))
```

Maybe there is something I did wrong, or a better way.

Thanks

Klikini, 2025-04-07:
Same problem with the TNC 640 simulator, but adding that copy loop after selecting an endpoint works, thanks!

magiconair, 2025-04-07:
Looks similar to #706 which was fixed in #779 which I've released today.

Klikini, 2025-04-07:
@magiconair I can confirm this is fixed in v0.7.4, thank you :)

closed 2025-04-07 by magiconair